RPi-Distro/pi-gen #130: Unable to install postfix during pi-gen stage3

RichardBronosky, 2017-10-25:
I have created a custom stage3 with

### 00-run.sh

```
#!/bin/bash -ex

# Preseed local packages See: https://wiki.debian.org/en/PackageManagement/Preseed https://coderwall.com/p/lryimq/postfix-silent-install-on-ubuntu
for file in $(ls files/*.deb 2>/dev/null || true); do
on_chroot << EOF
    debconf-set-selections <<<"postfix postfix/mailname string raspberrypi"
    debconf-set-selections <<<"postfix postfix/main_mailer_type string 'Internet Site'"
EOF
done

# vim: et sw=4 ts=4 sts=4 syntax=sh
```

### 01-packages-nr
```
postfix
```

### build-error.log
```
[16:01:56] Begin /home/ubuntu/ies-rpi/pi-gen/stage3/00-postfix/01-packages-nr
Reading package lists... Done
Building dependency tree
Reading state information... Done
The following additional packages will be installed:
  postfix-sqlite ssl-cert
Suggested packages:
  procmail postfix-mysql postfix-pgsql postfix-ldap postfix-pcre postfix-lmdb sasl2-bin dovecot-common postfix-cdb mail-reader ufw postfix-doc openssl-blacklist
The following NEW packages will be installed:
  postfix postfix-sqlite ssl-cert
0 upgraded, 3 newly installed, 0 to remove and 0 not upgraded.
Need to get 1691 kB of archives.
After this operation, 4009 kB of additional disk space will be used.
Get:1 http://mirrordirector.raspbian.org/raspbian stretch/main armhf ssl-cert all 1.0.39 [20.8 kB]
Get:2 http://mirrordirector.raspbian.org/raspbian stretch/main armhf postfix-sqlite armhf 3.1.6-0+deb9u1 [317 kB]
Get:3 http://mirrordirector.raspbian.org/raspbian stretch/main armhf postfix armhf 3.1.6-0+deb9u1 [1353 kB]
Fetched 1691 kB in 2s (812 kB/s)
qemu: Unsupported syscall: 384
Preconfiguring packages ...
Selecting previously unselected package ssl-cert.
(Reading database ... 34426 files and directories currently installed.)
Preparing to unpack .../ssl-cert_1.0.39_all.deb ...
Unpacking ssl-cert (1.0.39) ...
Selecting previously unselected package postfix-sqlite.
Preparing to unpack .../postfix-sqlite_3.1.6-0+deb9u1_armhf.deb ...
Unpacking postfix-sqlite (3.1.6-0+deb9u1) ...
Selecting previously unselected package postfix.
Preparing to unpack .../postfix_3.1.6-0+deb9u1_armhf.deb ...
Unpacking postfix (3.1.6-0+deb9u1) ...
Setting up ssl-cert (1.0.39) ...
Processing triggers for systemd (232-25+deb9u1) ...
Processing triggers for man-db (2.7.6.1-2) ...
Setting up postfix-sqlite (3.1.6-0+deb9u1) ...
/bin/grep: /etc/postfix/dynamicmaps.cf: No such file or directory
Adding sqlite map entry to /etc/postfix/dynamicmaps.cf
Processing triggers for rsyslog (8.24.0-1) ...
Running in chroot, ignoring request.
invoke-rc.d: policy-rc.d denied execution of try-restart.
Setting up postfix (3.1.6-0+deb9u1) ...
qemu: Unsupported syscall: 384
Created symlink /etc/systemd/system/multi-user.target.wants/postfix.service -> /lib/systemd/system/postfix.service.
Running in chroot, ignoring request.
invoke-rc.d: policy-rc.d denied execution of start.
Adding group `postfix' (GID 114) ...
Done.
Adding system user `postfix' (UID 109) ...
Adding new user `postfix' (UID 109) with group `postfix' ...
Not creating home directory `/var/spool/postfix'.
Adding group `postdrop' (GID 115) ...
Done.
/usr/sbin/postconf: fatal: inet_addr_local[getifaddrs]: getifaddrs: Address family not supported by protocol
/usr/sbin/postconf: fatal: inet_addr_local[getifaddrs]: getifaddrs: Address family not supported by protocol
dpkg: error processing package postfix (--configure):
 subprocess installed post-installation script returned error exit status 1
Processing triggers for systemd (232-25+deb9u1) ...
Processing triggers for rsyslog (8.24.0-1) ...
Running in chroot, ignoring request.
invoke-rc.d: policy-rc.d denied execution of try-restart.
Errors were encountered while processing:
 postfix
E: Sub-process /usr/bin/dpkg returned an error code (1)
```

XECDesign, 2017-10-27:
Just tried this out myself and it works fine here.
Are you building normally or using docker?

XECDesign, 2017-10-31:
No response

closed 2017-10-31 by XECDesign