RPi-Distro/pi-gen #133: failed to run command '/debootstrap/debootstrap': Exec format error

attie, 2017-11-13:
Trying to run on an x86-64 machine, running Ubuntu 16.04 with Docker 1.12.6, on commit 31fcc80.

My config:
```
$ cat config
IMG_NAME='my_project'
USE_QEMU=1
```

Interesting output is below:
```
$ ./build-docker.sh
Sending build context to Docker daemon 837.1 kB
Step 1 : FROM debian:stretch
 ---> 6d83de432e98
[...]
I: Extracting xz-utils...
I: Extracting zlib1g...
I: Running command: chroot /pi-gen/work/2017-11-13-4kopen/stage0/rootfs /debootstrap/debootstrap --second-stage
chroot: failed to run command '/debootstrap/debootstrap': Exec format error
rmdir: failed to remove '/pi-gen/work/2017-11-13-4kopen/stage0/rootfs/debootstrap': Directory not empty
```

I have confirmed that `qemu-debootstrap` is selected (according to `scripts/common:13`).

It looks like something is trying to run `/debootstrap/debootstrap` (a shell script) outside of QEMU - the script's shebang calls on `/bin/sh` which is a symlink to `/bin/dash` which is in turn an ARM binary (as below, from inside a container):
```
$ cd /pi-gen/work/2017-11-13-4kopen/stage0/rootfs
$ head -n 1 ./debootstrap/debootstrap
#!/bin/sh
$ ls -l ./bin/sh
lrwxrwxrwx 1 root root 4 Jan 24  2017 ./bin/sh -> dash
$ file ./bin/dash
./bin/dash: ELF 32-bit LSB executable, ARM, EABI5 version 1 (SYSV), dynamically linked, interpreter /lib/ld-linux-armhf.so.3, for GNU/Linux 3.2.0, BuildID[sha1]=18bddc2d67c785bbaaf97cbb2c0dd5420a1861b3, stripped
```

Can anyone help get this running?
Am I missing something?

Thanks!

attie, 2017-11-13:
Full output is below:
```
$ ./build-docker.sh
Sending build context to Docker daemon 837.1 kB
Step 1 : FROM debian:stretch
 ---> 6d83de432e98
Step 2 : ENV DEBIAN_FRONTEND noninteractive
 ---> Using cache
 ---> b212f8354864
Step 3 : RUN apt-get -y update &&     apt-get -y install         git vim parted         quilt realpath qemu-user-static debootstrap zerofree pxz zip dosfstools         bsdtar libcap2-bin rsync grep udev xz-utils curl xxd     && rm -rf /var/lib/apt/lists/*
 ---> Using cache
 ---> 77833594a8b1
Step 4 : COPY . /pi-gen/
 ---> Using cache
 ---> 00f62df9f876
Step 5 : VOLUME /pi-gen/work /pi-gen/deploy
 ---> Using cache
 ---> af7c94e54f29
Successfully built af7c94e54f29
update-binfmts: warning: Couldn't load the binfmt_misc module.
update-binfmts: warning: Couldn't load the binfmt_misc module.
update-binfmts: warning: Couldn't load the binfmt_misc module.
update-binfmts: warning: Couldn't load the binfmt_misc module.
update-binfmts: warning: Couldn't load the binfmt_misc module.
update-binfmts: warning: Couldn't load the binfmt_misc module.
update-binfmts: warning: Couldn't load the binfmt_misc module.
update-binfmts: warning: Couldn't load the binfmt_misc module.
update-binfmts: warning: Couldn't load the binfmt_misc module.
update-binfmts: warning: Couldn't load the binfmt_misc module.
update-binfmts: warning: Couldn't load the binfmt_misc module.
update-binfmts: warning: Couldn't load the binfmt_misc module.
update-binfmts: warning: Couldn't load the binfmt_misc module.
update-binfmts: warning: Couldn't load the binfmt_misc module.
update-binfmts: warning: Couldn't load the binfmt_misc module.
update-binfmts: warning: Couldn't load the binfmt_misc module.
update-binfmts: warning: Couldn't load the binfmt_misc module.
update-binfmts: warning: Couldn't load the binfmt_misc module.
update-binfmts: warning: Couldn't load the binfmt_misc module.
update-binfmts: warning: Couldn't load the binfmt_misc module.
update-binfmts: warning: Couldn't load the binfmt_misc module.
[16:04:44] Begin /pi-gen
[16:04:44] Begin /pi-gen/stage0
[16:04:44] Begin /pi-gen/stage0/prerun.sh
I: Running command: debootstrap --arch armhf --foreign --components=main,contrib,non-free --keyring /pi-gen/stage0/files/raspberrypi.gpg stretch /pi-gen/work/2017-11-13-4kopen/stage0/rootfs http://mirrordirector.raspbian.org/raspbian/
I: Retrieving InRelease
I: Checking Release signature
I: Valid Release signature (key id A0DA38D0D76E8B5D638872819165938D90FDDD2E)
I: Retrieving Packages
I: Validating Packages
I: Retrieving Packages
I: Validating Packages
I: Retrieving Packages
I: Validating Packages
I: Resolving dependencies of required packages...
I: Resolving dependencies of base packages...
I: Found additional required dependencies: adduser cpio dmsetup initramfs-tools initramfs-tools-core klibc-utils kmod libapparmor1 libaudit-common libaudit1 libbz2-1.0 libcap-ng0 libcap2 libcryptsetup4 libdb5.3 libdbus-1-3 libdebconfclient0 libdevmapper1.02.1 libdrm2 libgcrypt20 libgpg-error0 libidn11 libip4tc0 libklibc libkmod2 liblz4-1 libncursesw5 libnih-dbus1 libnih1 libpcre3 libplymouth4 libpng16-16 libprocps6 libseccomp2 libsemanage-common libsemanage1 libsystemd0 libudev1 libustr-1.0-1 linux-base makedev plymouth procps systemd udev
I: Found additional base dependencies: aptitude-common gnupg gnupg-agent libassuan0 libboost-filesystem1.62.0 libboost-iostreams1.62.0 libboost-system1.62.0 libbsd0 libcwidget3v5 libdns-export162 libelf1 libestr0 libfastjson4 libffi6 libgmp10 libgnutls30 libhogweed4 libidn2-0 libip6tc0 libiptc0 libisc-export160 libksba8 liblogging-stdlog0 liblognorm5 libnetfilter-conntrack3 libnettle6 libnfnetlink0 libnpth0 libp11-kit0 libpsl5 libsigc++-2.0-0v5 libsqlite3-0 libtasn1-6 libunistring0 libxtables12 pinentry-curses xxd
I: Checking component main on http://mirrordirector.raspbian.org/raspbian...
I: Retrieving adduser 3.115
I: Validating adduser 3.115
I: Retrieving apt 1.4.8
I: Validating apt 1.4.8
I: Retrieving apt-utils 1.4.8
I: Validating apt-utils 1.4.8
I: Retrieving aptitude 0.8.7-1
I: Validating aptitude 0.8.7-1
I: Retrieving aptitude-common 0.8.7-1
I: Validating aptitude-common 0.8.7-1
I: Retrieving base-files 9.9+rpi1+deb9u1
I: Validating base-files 9.9+rpi1+deb9u1
[... 499 lines cut ...]
I: Extracting makedev...
I: Extracting mawk...
I: Extracting mount...
I: Extracting mountall...
I: Extracting multiarch-support...
I: Extracting ncurses-base...
I: Extracting ncurses-bin...
I: Extracting passwd...
I: Extracting perl-base...
I: Extracting plymouth...
I: Extracting procps...
I: Extracting sed...
I: Extracting sensible-utils...
I: Extracting systemd...
I: Extracting sysvinit-utils...
I: Extracting tar...
I: Extracting tzdata...
I: Extracting udev...
I: Extracting util-linux...
I: Extracting xz-utils...
I: Extracting zlib1g...
I: Running command: chroot /pi-gen/work/2017-11-13-4kopen/stage0/rootfs /debootstrap/debootstrap --second-stage
chroot: failed to run command '/debootstrap/debootstrap': Exec format error
rmdir: failed to remove '/pi-gen/work/2017-11-13-4kopen/stage0/rootfs/debootstrap': Directory not empty
```

XECDesign, 2017-11-13:
Try to figure out why you're seeing `update-binfmts: warning: Couldn't load the binfmt_misc module.`. That's what's leading to the problem.

attie, 2017-11-13:
Thanks very much! I did wonder about them.
For future reference, I ran `modprobe binfmt_misc` on the host, and re-ran `./build-docker.sh`... I suspect the container couldn't insert the module?
It's running now 👍

closed 2017-11-13 by attie

tkurki, 2018-02-10:
Thanks for sharing this, bumped against the same error and the same fix worked. 

Should this be added to the README or a separate TROUBLESHOOTING doc?

Fmstrat, 2018-07-09:
Agreed. Should be added to documentation.

ProtoSD, 2019-12-26:
Experienced this error with a Docker debian:buster container. It took some head banging, but in case anyone else runs into this, the fix for me was to run the container with --privileged
`docker run -it --privileged debian:buster /bin/bash`

rst0git, 2021-06-25:
> Try to figure out why you're seeing update-binfmts: warning: Couldn't load the binfmt_misc module.. That's what's leading to the problem.

I had the same problem and running the follow command before `./build-docker.sh` resolved it.
```
docker run --rm --privileged multiarch/qemu-user-static --reset -p yes
```