RPi-Distro/pi-gen #134: unbootable images

wintermute101, 2017-11-16:
On dev branch in file:
export-image/03-set-partuuid/00-run.sh

there is xxd -e

xxd command on my system have no -e switch (typo?) it silently fails and produce unbootable build
to fix it I reverted to previous version

XECDesign, 2017-11-16:
What does `xxd -v` return?

wintermute101, 2017-11-17:
xxd -v
xxd V1.10 27oct98 by Juergen Weigert

xxd -h
Usage:
       xxd [options] [infile [outfile]]
    or
       xxd -r [-s [-]offset] [-c cols] [-ps] [infile [outfile]]
Options:
    -a          toggle autoskip: A single '*' replaces nul-lines. Default off.
    -b          binary digit dump (incompatible with -ps,-i,-r). Default hex.
    -c cols     format <cols> octets per line. Default 16 (-i: 12, -ps: 30).
    -E          show characters in EBCDIC. Default ASCII.
    -g          number of octets per group in normal output. Default 2.
    -h          print this summary.
    -i          output in C include file style.
    -l len      stop after <len> octets.
    -ps         output in postscript plain hexdump style.
    -r          reverse operation: convert (or patch) hexdump into binary.
    -r -s off   revert with <off> added to file positions found in hexdump.
    -s [+][-]seek  start at <seek> bytes abs. (or +: rel.) infile offset.
    -u          use upper case hex letters.
    -v          show version: "xxd V1.10 27oct98 by Juergen Weigert".

XECDesign, 2017-11-17:
What about `apt-cache policy xxd`?
```
$ xxd -v
xxd V1.10 27oct98 by Juergen Weigert
$ xxd -h
Usage:
       xxd [options] [infile [outfile]]
    or
       xxd -r [-s [-]offset] [-c cols] [-ps] [infile [outfile]]
Options:
    -a          toggle autoskip: A single '*' replaces nul-lines. Default off.
    -b          binary digit dump (incompatible with -ps,-i,-r). Default hex.
    -c cols     format <cols> octets per line. Default 16 (-i: 12, -ps: 30).
    -E          show characters in EBCDIC. Default ASCII.
    -e          little-endian dump (incompatible with -ps,-i,-r).
    -g          number of octets per group in normal output. Default 2 (-e: 4).
    -h          print this summary.
    -i          output in C include file style.
    -l len      stop after <len> octets.
    -o off      add <off> to the displayed file position.
    -ps         output in postscript plain hexdump style.
    -r          reverse operation: convert (or patch) hexdump into binary.
    -r -s off   revert with <off> added to file positions found in hexdump.
    -s [+][-]seek  start at <seek> bytes abs. (or +: rel.) infile offset.
    -u          use upper case hex letters.
    -v          show version: "xxd V1.10 27oct98 by Juergen Weigert".
```

wintermute101, 2017-11-20:
Please note that host system is Ubuntu 14.04 not Debian.

$ apt-cache policy xxd
N: Unable to locate package xxd

$which xxd
/usr/bin/xxd

$|dpkg --search /usr/bin/xxd
vim-common: /usr/bin/xxd

On ubuntu 16.04

$xxd -h
Usage:
       xxd [options] [infile [outfile]]
    or
       xxd -r [-s [-]offset] [-c cols] [-ps] [infile [outfile]]
Options:
    -a          toggle autoskip: A single '*' replaces nul-lines. Default off.
    -b          binary digit dump (incompatible with -ps,-i,-r). Default hex.
    -c cols     format <cols> octets per line. Default 16 (-i: 12, -ps: 30).
    -E          show characters in EBCDIC. Default ASCII.
    -e          little-endian dump (incompatible with -ps,-i,-r).
    -g          number of octets per group in normal output. Default 2 (-e: 4).
    -h          print this summary.
    -i          output in C include file style.
    -l len      stop after <len> octets.
    -o off      add <off> to the displayed file position.
    -ps         output in postscript plain hexdump style.
    -r          reverse operation: convert (or patch) hexdump into binary.
    -r -s off   revert with <off> added to file positions found in hexdump.
    -s [+][-]seek  start at <seek> bytes abs. (or +: rel.) infile offset.
    -u          use upper case hex letters.
    -v          show version: "xxd V1.10 27oct98 by Juergen Weigert".
$ xxd -v
xxd V1.10 27oct98 by Juergen Weigert
$ dpkg --search /usr/bin/xxd
vim-common: /usr/bin/xxd

Seems like xxd version was not updated :(

$apt-cache show vim-common
Package: vim-common
Architecture: amd64
Version: 2:7.4.052-1ubuntu3.1

$ apt-cache  show vim-common
Package: vim-common
Architecture: amd64
Version: 2:7.4.1689-3ubuntu1.2

XECDesign, 2017-11-20:
It's difficult to support every debian/ubuntu derivative and version due to these subtle changes in features and behaviour of various tools, so I'll have to close this one. Currently, stretch and xenial are supported.

closed 2017-11-20 by XECDesign