RPi-Distro/pi-gen #140: build fails in Jessie

Equidamoid, 2017-12-29:
Due to https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=857338 debootstrap fails with the following error in debootstrap.log:
```
dpkg: regarding .../archives/init_1.48_armhf.deb containing init, pre-dependency problem:
 init pre-depends on systemd-sysv | sysvinit-core
  systemd-sysv is unpacked, but has never been configured.
  sysvinit-core is not installed.

dpkg: error processing archive /var/cache/apt/archives/init_1.48_armhf.deb (--unpack):
 pre-dependency problem - not installing init
```

I understand that it's an upstream issue, but maybe it is worth adding a version check to build.sh.

In case someone else encounters this problem, the solution is to install debootstrap from jessie-backports: add `deb http://httpredir.debian.org/debian jessie-backports main` to your `sources.list` and then run `apt update; apt -t jessie-backports install debootstrap`.

XECDesign, 2018-01-24:
I've run into this issue myself ages ago, so it would probably make sense to check for it and avoid wasting peoples' time. I'd accept a PR.

XECDesign, 2018-03-01:
No activity

closed 2018-03-01 by XECDesign