RPi-Distro/pi-gen #143: Updating custom build using `CLEAN=1` fails with `Previous stage rootfs not found`

andig, 2018-01-01:
I'm trying to develop a custom build using the docker infrastructure:

  - removed stages 3-5
  - removed stage2/EXPORT*
  - added custom stage 3

Initial build ran fine:

	copying results from deploy/
	total 866288
	drwxr-xr-x   5 andig  staff   160B 31 Dec 16:46 .
	drwxr-xr-x  26 andig  staff   832B 31 Dec 15:37 ..
	-rw-r--r--   1 andig  staff    60K 31 Dec 16:46 2017-12-31-Raspbian-volkszaehler.info
	-rw-r--r--   1 andig  staff   6.0K 31 Dec 16:46 build.log
	-rw-r--r--   1 andig  staff   420M 31 Dec 16:46 image_2017-12-31-Raspbian-volkszaehler.zip
	pigen_work
	Done! Your image(s) should be in deploy/

Now I've added `SKIP` to stages 0..2, made changes to stage 3 and re-ran:

	❯ touch stage2/SKIP
	❯ touch stage1/SKIP
	❯ touch stage0/SKIP

	❯ CLEAN=1 ./build-docker.sh
	Sending build context to Docker daemon  3.002MB
	Step 1/5 : FROM debian:stretch
	 ---> da653cee0545
	Step 2/5 : ENV DEBIAN_FRONTEND noninteractive
	 ---> Using cache
	 ---> aca0071a24c5
	Step 3/5 : RUN apt-get -y update &&     apt-get -y install         git vim parted         quilt realpath qemu-user-static debootstrap zerofree pxz zip dosfstools         bsdtar libcap2-bin rsync grep udev xz-utils curl xxd php7.0     && rm -rf /var/lib/apt/lists/*
	 ---> Using cache
	 ---> ebcbbfec4a21
	Step 4/5 : COPY . /pi-gen/
	 ---> 298996574d76
	Step 5/5 : VOLUME /pi-gen/work /pi-gen/deploy
	 ---> Running in 1f76a5ca87ba
	 ---> c2da159b6b6e
	Removing intermediate container 1f76a5ca87ba
	Successfully built c2da159b6b6e
	Successfully tagged pi-gen:latest
	update-binfmts: warning: qemu-aarch64 already enabled in kernel.
	[17:05:20] Begin /pi-gen
	[17:05:20] Begin /pi-gen/stage0
	[17:05:20] End /pi-gen/stage0
	[17:05:20] Begin /pi-gen/stage1
	[17:05:20] End /pi-gen/stage1
	[17:05:20] Begin /pi-gen/stage2
	[17:05:20] End /pi-gen/stage2
	[17:05:20] Begin /pi-gen/stage3
	[17:05:20] Begin /pi-gen/stage3/prerun.sh
	Previous stage rootfs not found

Is this potentially because the container has been updated meanwhile (I have a `prepare.sh` script in `.` that I'm changing)? If not- what else could lead to rootfs not being found when tweaking an existing build?

XECDesign, 2018-01-01:
Afraid I can't help with Docker builds, since that has been provided through pull requests and I don't use it myself.

andig, 2018-01-02:
ping @cfstras would this be something you could kindly help with?

andig, 2018-01-02:
It seems to me that stage2 is doing something strange. Steps to repeat:

  - remove stages 3-5
  - touch `stage2/SKIP`, add `stage1/SKIP`
  - run build and inspect result:

        ❯ docker run -it --rm --privileged \
                        --volumes-from="${CONTAINER_NAME}" --name "${CONTAINER_NAME}_cont" \
                        -e IMG_NAME=${IMG_NAME} \
                        pi-gen \
                        bash

        root@37dbce45db89:/# ls pi-gen/work/
        2018-01-02-Raspbian

        root@37dbce45db89:/# ls pi-gen/work/2018-01-02-Raspbian/
        build.log  export-image  stage0

        root@37dbce45db89:/# ls pi-gen/work/2018-01-02-Raspbian/stage0/rootfs/
        bin  boot  dev  etc  home  lib  media  mnt  opt  proc  root  run  sbin  srv  sys  tmp  usr  var

  - now touch `stage0/SKIP`
  - run build and inspect result:

        ❯ CONTINUE=1 ./build-docker.sh
        ❯ docker run -it --rm --privileged \
                        --volumes-from="${CONTAINER_NAME}" --name "${CONTAINER_NAME}_cont" \
                        -e IMG_NAME=${IMG_NAME} \
                        pi-gen \
                        bash
        
        root@3357542e3ea8:/# cd pi-gen/
        root@3357542e3ea8:/pi-gen# ls work/2018-01-02-Raspbian/
        build.log  export-image  stage0  stage1
        root@3357542e3ea8:/pi-gen# ls work/2018-01-02-Raspbian/stage0/rootfs/
        bin  boot  dev  etc  home  lib  media  mnt  opt  proc  root  run  sbin  srv  sys  tmp  usr  var
        root@3357542e3ea8:/pi-gen# ls work/2018-01-02-Raspbian/stage1/rootfs/
        bin  boot  dev  etc  home  lib  media  mnt  opt  proc  root  run  sbin  srv  sys  tmp  usr  var

`rootfs` of both stages is present. Now try to build stage2 in addition:

  - now touch `stage1/SKIP`
  - run build and inspect result:

        ❯ CONTINUE=1 ./build-docker.sh
        ❯ docker run -it --rm --privileged \
                        --volumes-from="${CONTAINER_NAME}" --name "${CONTAINER_NAME}_cont" \
                        -e IMG_NAME=${IMG_NAME} \
                        pi-gen \
                        bash
        root@4111af6747d6:/# ls pi-gen/work/
        2018-01-02-Raspbian
        root@4111af6747d6:/# ls pi-gen/work/2018-01-02-Raspbian/
        build.log

Suddenly the work dir is empty. I'll try to build stages 0+1 once more, preserve the result and do some further testing what happens in stage2.

andig, 2018-01-02:
I guess I've found it: https://github.com/RPi-Distro/pi-gen/blob/dev/build-docker.sh#L78

After successful build the docker container is removed, including its working directory and rootfs.

XECDesign, 2018-01-02:
Can this be closed with https://github.com/RPi-Distro/pi-gen/pull/145?

andig, 2018-01-03:
> Can this be closed with #145?

This is different. Please allow me to spend a little more time testing to confirm my findings before making a proposal.

cfstras, 2018-01-05:
@andig: you're right. You could temporarily comment out that line, or add a config option `CLEANUP_WORK=1 (default)` to disable the behavior.

andig, 2018-01-05:
> You could temporarily comment out that line, or add a config option CLEANUP_WORK=1 (default) to disable the behavior.

Yes. I'd like to keep this open until we have a PR whoever is first ;)

closed 2018-03-05 by XECDesign