RPi-Distro/pi-gen #146: What is on_chroot used for?

spuder, 2018-01-04:
Looking at the scripts, I see some commands are wrapped in this command

```
on_chroot << EOF

EOF
```

I can't tell what the reasoning is, why do some commands need to be wrapped in on_chroot, but others don't?

m-roberts, 2018-01-05:
A quick search of the project will show you that this function comes from [scripts/common](https://github.com/RPi-Distro/pi-gen/blob/dev/scripts/common#L71).

The key line here is `capsh --drop=cap_setfcap "--chroot=${ROOTFS_DIR}/" -- "$@"`, where `"$@"` is whatever is 'wrapped' in on_chroot.

`--chroot=${ROOTFS_DIR}/` means that the command is run **as if it were on the Pi**, rather than on the host machine.

spuder, 2018-01-05:
Thanks for the information. I've done some more testing and I see that without `on_chroot` I get 'directory not found errors'. This makes sense now.

closed 2018-01-05 by spuder