RPi-Distro/pi-gen #149: Feature request: docker-build.sh argument to export/not export image

jraviotta, 2018-01-17:
Similar to [Feature Request: Modular Builds](https://github.com/RPi-Distro/pi-gen/issues), I want to control the export of the image via an argument on build_docker.sh instead of editing the contents of the repository.

This is important to me as i (and my dev team) want to be able to follow the git branch > merge workflow without having to deal with deployment changes that differ across branches. 

I am willing to help with a pr but it would require some re-architecture of the build process that should be considered by the project owners.

XECDesign, 2018-01-24:
There are definitely more sensible ways of doing it than `SKIP`. I'd add argument parsing to `build.sh` and pass arguments from `docker-build.sh` to `build.sh`. There's no time to work on pi-gen right now, so it would have to be a PR.

I am not a fan of tools which have so many arguments that you have to refer to the manpage each time you want to use it, so I'd prefer if there was only a handful. Perhaps controlling whether it's a clean build, whether exports should be skipped and an option to override the config file. Then everything else should be tweaked through the config file.

Are there other arguments which would be good to have in addition to those listed above?

jraviotta, 2018-01-24:
Good point on keeping command line arguments simple. How about a .yml
config file for RPi settings? Then the scripts could refer to variables
instead of hardcoded values?

XECDesign, 2018-01-24:
What advantage does YAML give over the current config file approach?

If there are good reasons to add more structure, why not json instead? If the idea, in the long term, is to move from bash to python, I'd prefer json, since we already use it heavily in other tools.

samtygier, 2019-02-16:
Note that passing arguments from `docker-build.sh` through to `build.sh` is included in #259