RPi-Distro/pi-gen #158: "Operation not permitted" in stage0/prerun.sh

spikespaz, 2018-02-18:
vagrant@ubuntu-xenial:/vagrant$ sudo ./build.sh
    [04:11:35] Begin /vagrant
    [04:11:35] Begin /vagrant/stage0
    [04:11:36] Begin /vagrant/stage0/prerun.sh
    I: Running command: debootstrap --arch armhf --foreign --components=main,contrib,non-free --keyring /vagrant/stage0/files/raspberrypi.gpg stretch /vagrant/work/2018-02-18-Raspbian/stage0/rootfs http://mirrordirector.raspbian.org/raspbian/
    mknod: /vagrant/work/2018-02-18-Raspbian/stage0/rootfs/test-dev-null: Operation not permitted
    E: Cannot install into target '/vagrant/work/2018-02-18-Raspbian/stage0/rootfs' mounted with noexec or nodev
    rmdir: failed to remove '/vagrant/work/2018-02-18-Raspbian/stage0/rootfs/debootstrap': No such file or directory

XECDesign, 2018-03-01:
@spikespaz Any luck figuring this out? I think the "mounted with noexec or nodev" is a big hint.

spikespaz, 2018-03-01:
@XECDesign No, I am a complete novice at this.

XECDesign, 2018-03-01:
Are you building on an NFS mount? Could you post the output of 'mount'?

conradek, 2018-03-27:
Could the issue be an encrypted filesystem?

XECDesign, 2018-03-27:
The issue is that the filesystem is `mounted with noexec or nodev`. Without knowing OP's setup, there's not much that can be done here. It's not a pi-gen issue and there has been no response in 26 days, so it's probably time to close this one.

closed 2018-03-27 by XECDesign

spikespaz, 2018-04-06:
Apologies for the (extremely) late response. Apparently my spam folder was eating all of these messages, and I just noticed while checking my GitHub notifications. I was trying to do this inside of a Vagrant virtual machine on windows using the VirtualBox provider. It was using the latest Xenial box. I stopped using Vagrant because of the questionable quality of VirtualBox, and I am now going to attempt this again in an Ubuntu Server Xenial environment inside of VMWare Player.