RPi-Distro/pi-gen #161: Image not booting

moritzvieli, 2018-03-11:
Hi there!

I haven't found a solution nor an indication where to look for a detailed error, so I'm posting my issue here: I built an own lite-image and changed nothing else from tag2017-11-29-raspbian-stretch. The built went through and I've not seen any error in the logs.

After flashing my image to the SD card, only the raspberry images are shown, but the boot process stops. No text is written, the cursor just flashes. I can enter characters, but they don't seem to change anything.

When I flash the official raspbian lite image, everything works fine.

Do you have an idea, where I could get an error message or do you have a clue, what went wrong?

Many thanks,
Moritz

XECDesign, 2018-03-11:
Is it any different if you build from master?

ibs92, 2018-03-11:
I am having the exact same issue... Tried different tags and both master and dev branches but with the same result.

ibs92, 2018-03-11:
I solved the issue I was having: 

I was using Debian jessie running an raspberry pi. Installed Debian stretch for amd64 on a virtual machine, generated the image and it worked without any problems. I guess there is some sort of problem using  Debian jessie (even that you don't get any error when generating the image)

XECDesign, 2018-03-12:
Thanks, @ibs92.

@moritzvieli What OS are you using?

moritzvieli, 2018-03-12:
That's it! Thanks a lot @ibs92! I had a Jessie version running and tried again now with the latest version of a Debian Stretch and it worked without a problem! You saved my day! :)

closed 2018-03-12 by moritzvieli