RPi-Distro/pi-gen #162: how do we lock down package versions?

monkey-jsun, 2018-03-12:
Maybe this is a stupid question - does pi-gen always fetch the latest versions of packages?

A little more background.  For a production system, we will create release branch for pi-gen.  Once in a (hopefully) long while we might fix a bug somewhere and re-spin the build.  What we really want is that during the re-spin, all packages would still use the same versions as all previous build in the same release branch, instead of latest versions, because system was well tested and we don't want any surprises.

Is this possible?

XECDesign, 2018-03-12:
One clunky approach that would require the least work:

Use `apt-cacher-ng` and the `APT_PROXY` parameter.
Once the build finishes, copy the cached files into a snapshot directory.
When you want to rebuild from that snapshot, use a web server pointing to that snapshot directory as the repos you're installing from.

I haven't tried it, so I can't say that it will definitely work.

Another approach would be to use `debmirror` instead of `apt-cacher-ng`, but that would download all the packages rather than just the ones used for your build.

Landrash, 2018-03-14:
The solutions suggested by @XECDesign are sound and would most likely be the recommended approach since you most likely need/want to keep a copy of all the packages anyway for a production system.

monkey-jsun, 2018-03-14:
@XECDesign  What do you mean by "use a web server pointing to that snapshot directory as the repos you're installing from"?  I would assume all packages are represented as some URL's.  Are you saying I need to hack the HTTP client used by py-getn to fetch from local directory?  Clearly I can't hack the servers in those URL's.

@Landrash  Are you referring to @XECDesign  #1 or #2 solution?

Thanks to you both!

XECDesign, 2018-03-14:
You can configure apt-cacher-ng or a proxy of your choice to redirect requests for raspberrypi.org and raspbian.org to your own mirror. Or you can modify /etc/apt/sources.list{.d/raspi.list} to point to your mirror.

XECDesign, 2018-03-28:
No activity

closed 2018-03-28 by XECDesign