RPi-Distro/pi-gen #164: "No space left on device"

davesteele, 2018-03-19:
I had an image that failed to resize on first boot, and failed pretty much everything else after that, with a full root filesystem. It looks like the amount of free space is luck of the draw.

It wasn't clear to me where to add some default space in export-image/prerun.sh.

This workaround worked (doesn't consider NOOBS):

    diff --git a/export-image/04-finalise/01-run.sh b/export-image/04-finalise/01-run.sh
    index b706774..e82c8df 100755
    --- a/export-image/04-finalise/01-run.sh
    +++ b/export-image/04-finalise/01-run.sh
    @@ -39,6 +39,8 @@ find "${ROOTFS_DIR}/var/log/" -type f -exec cp /dev/null {} \;
     rm -f "${ROOTFS_DIR}/root/.vnc/private.key"
     rm -f "${ROOTFS_DIR}/etc/vnc/updateid"
 
    +rm -f "${ROOTFS_DIR}/home/pi/waste"
    +
     update_issue "$(basename "${EXPORT_DIR}")"
     install -m 644 "${ROOTFS_DIR}/etc/rpi-issue" "${ROOTFS_DIR}/boot/issue.txt"
     install files/LICENSE.oracle "${ROOTFS_DIR}/boot/"
    diff --git a/stage2/04-waste-space/00-run.sh b/stage2/04-waste-space/00-run.sh
    new file mode 100755
    index 0000000..95f5b8e
    --- /dev/null
    +++ b/stage2/04-waste-space/00-run.sh
    @@ -0,0 +1,4 @@
    +#!/bin/bash
    +
    +dd if=/dev/zero of=${ROOTFS_DIR}/home/pi/waste bs=1M count=100
    +

XECDesign, 2018-03-19:
It seems like this would be equivalent to adding more here: https://github.com/RPi-Distro/pi-gen/blob/master/export-image/prerun.sh#L17

I haven't found a good solution that doen't involve just constantly bumping that number. The output du gives isn't reliable, since it depends on the source filesystem. Maybe we should make a larger image and then shrink it down. rather than try to guess how much space it needs. That doesn't work too well either, since ext4 filesystems will want to reserve some percentage of the free space and refuse to resize below that.

davesteele, 2018-03-19:
My previous solution used the post shrink strategy - I don't recommend it. Running resize2fs based on the free space was unreliable, and resulted in larger images than you might expect. My ultimate strategy here was to loop over resize, starting with a target of ~0bytes, until it passed.

The good news is that the extra zeroed space compresses down to nothing, and adds little to SD size requirements and burn time.

Perhaps, as a start, FREE_SPACE should be available to all exports. (a3d1488).

davesteele, 2018-03-20:
Well, it wasn't free space.

The patch to add init_resize to /boot/cmdline.txt wasn't applied for some reason. Applying it manually worked.

whatever is going on, it has nothing to do with the contents of this thread. Closing.

closed 2018-03-20 by davesteele