RPi-Distro/pi-gen #166: Previous stage rootfs not found

cburlacu, 2018-03-19:
One cannot skip an intermediary stage.
One possible solution is to move the line
```PREV_ROOTFS_DIR="${ROOTFS_DIR}"```
inside the ```if [ ! -f SKIP ] ``` statement

fdaugs, 2018-03-28:
I had the same problem. My fix was to rename the latest folder in the 'work' directory to the current date. So the issue was probably caused by the script searching for a folder with the current date. I'm not sure, wether it is an intended behavior.

xoronet, 2018-04-06:
Unfortunately the same problem is described in #176

david-kohen, 2018-04-26:
See my comment in  #176 about adding a stage2a

rtgoodwin, 2018-10-24:
As a newb to the project, the find-rootfs-by-date mechanism definitely bites you when working with your own stages (to keep future base merges clean) and it takes you multiple days to get something done. (Right now it always takes me at least a couple of days, and several iterations, since I'm still learning. 

Dozens of ways to fix it, just voting for some option. Even something as simple as adding an optional env var in the config for "IMG_BUILD_DATE" or similar to tell copy_previous where to look for rootfs. If that's something that would be accepted I can contribute that.

XECDesign, 2018-10-24:
A stupid workaround I use is soemthing like "IMG_DATE=2017-03-02", which should work.

rtgoodwin, 2018-10-24:
D'oh. Didn't even read that bit of `build` to notice that. :)

andig, 2019-03-17:
@XECDesign just hit the same stopper. Problem is that the `IMG_DATE` does't carry into the docker build:

    CONTINUE=1 PRESERVE_CONTAINER=1 IMG_DATE=2019-03-11 ./build-docker.sh

gives:

    [16:53:08] Begin /pi-gen
    [16:53:08] Begin /pi-gen/stage0
    [16:53:08] End /pi-gen/stage0
    [16:53:08] Begin /pi-gen/stage1
    [16:53:08] End /pi-gen/stage1
    [16:53:08] Begin /pi-gen/stage2
    [16:53:08] End /pi-gen/stage2
    [16:53:08] Begin /pi-gen/volkszaehler
    [16:53:08] Begin /pi-gen/volkszaehler/prerun.sh
    >> /pi-gen/work/2019-03-17-volkszaehler/stage2/rootfs
    Previous stage rootfs not found