RPi-Distro/pi-gen #174: Unable to locate packages

breadam, 2018-03-28:
It was building fine a couple days ago, but now, I'm getting these two errors:

```
E: Unable to locate package wpasupplicant
E: Unable to locate package wireless-tools
```
I know this not a problem with pi-gen but i have no idea how to solve this. 
Thanks for any suggestions,

XECDesign, 2018-03-28:
what's the output if you run `apt-cache policy` in the chroot?

conradek, 2018-03-28:
I too am running into this issue.  apt-cache policy in chroot reports:

Package files:
 100 /var/lib/dpkg/status
     release a=now
 500 http://archive.raspberrypi.org/debian stretch/ui armhf Packages
     release o=Raspberry Pi Foundation,a=stable,n=stretch,l=Raspberry Pi Foundation,c=ui,b=armhf
     origin archive.raspberrypi.org
 500 http://archive.raspberrypi.org/debian stretch/main armhf Packages
     release o=Raspberry Pi Foundation,a=stable,n=stretch,l=Raspberry Pi Foundation,c=main,b=armhf
     origin archive.raspberrypi.org
Pinned packages:

softerra, 2018-03-29:
Hi All,
The same issue is observed.
The error happens on package installation at stage2/02-net-tweaks

- *apt-cache policy* in chroot of **stage2** - gives the output mentioned by @conradek 
- *apt-cache policy wpasupplicant* reports:
```
N: Unable to locate package wpasupplicant
```
It looks like some package of the ones installed before stage2/02-net-tweaks clears apt cache

- *apt-cache policy* in chroot of **stage1** - gives better result:
```
Package files:
 100 /var/lib/dpkg/status
     release a=now
 500 http://archive.raspberrypi.org/debian stretch/ui armhf Packages
     release o=Raspberry Pi Foundation,a=stable,n=stretch,l=Raspberry Pi Foundation,c=ui,b=armhf
     origin archive.raspberrypi.org
 500 http://archive.raspberrypi.org/debian stretch/main armhf Packages
     release o=Raspberry Pi Foundation,a=stable,n=stretch,l=Raspberry Pi Foundation,c=main,b=armhf
     origin archive.raspberrypi.org
 500 http://mirrordirector.raspbian.org/raspbian stretch/rpi armhf Packages
     release o=Raspbian,a=stable,n=stretch,l=Raspbian,c=rpi,b=armhf
     origin mirrordirector.raspbian.org
 500 http://mirrordirector.raspbian.org/raspbian stretch/non-free armhf Packages
     release o=Raspbian,a=stable,n=stretch,l=Raspbian,c=non-free,b=armhf
     origin mirrordirector.raspbian.org
 500 http://mirrordirector.raspbian.org/raspbian stretch/contrib armhf Packages
     release o=Raspbian,a=stable,n=stretch,l=Raspbian,c=contrib,b=armhf
     origin mirrordirector.raspbian.org
 500 http://mirrordirector.raspbian.org/raspbian stretch/main armhf Packages
     release o=Raspbian,a=stable,n=stretch,l=Raspbian,c=main,b=armhf
     origin mirrordirector.raspbian.org
Pinned packages:
```
- *apt-cache policy wpasupplicant* in the chroot of **stage1** reports:
```
wpasupplicant:
  Installed: (none)
  Candidate: 2:2.4-1+deb9u1
  Version table:
     2:2.4-1+deb9u1 500
        500 http://mirrordirector.raspbian.org/raspbian stretch/main armhf Packages
```

softerra, 2018-03-29:
Tried installing packages from stage2/01-sys-tweaks/00-packages one by one with verification the state of apt-policy after every installation.
According to the logs gotten, the failure begings after
```
...
Setting up gettext-base (0.19.8.1-2) ...
Processing triggers for man-db (2.7.6.1-2) ...
Setting up rfkill (0.5-1) ...
Setting up raspberrypi-sys-mods (20180328+1) ...
Fixing file capabilities...
Failed to set capabilities on file `/bin/ping6' (Invalid argument)
The value of the capability argument is not permitted for a file. Or the file is not a regular (non-symlink) file
Adjusting cmdline.txt...
Removing old sysctl tweaks, if present...
Removing old NOPASSWD line, if present...
Updating documentation URL in /boot/config.txt...
Updating apt sources from {mirrordirector,archive}.raspbian.org to raspbian.raspberrypi.org...
...
E: Unable to locate package apt-listchanges
...
```
It seems to happen because of the changes of 2018-03-28 in *raspberrypi-sys-mods*, doesn't it?

softerra, 2018-03-29:
cherry-pick of the latest commit https://github.com/RPi-Distro/pi-gen/commit/93db3736721d968a6ab67d37243d6f3aa6958d00 into your branch fixes the problem

XECDesign, 2018-03-29:
Into our branch? What branch? If you're building off 'dev', you shouldn't be.

breadam, 2018-03-29:
Hello again, 

Sorry for my late response, I woke up like two hours ago. Haven't been sleeping for like three days

I cloned the repo again and rebuilt, that solved my problem. I still don't know what the root cause is. I'm not really good at linux stuff. 

Thanks for your help @XECDesign and @softerra, 

Should I close this issue or keep it alive? Don't know

Have a good day :)

XECDesign, 2018-03-29:
Thanks for the update.

I suspect you were re-using a stage built before https://github.com/RPi-Distro/pi-gen/commit/93db3736721d968a6ab67d37243d6f3aa6958d00 or were working off the wrong branch. If your issues is resolved now, I'm inclined to close it.

closed 2018-03-29 by XECDesign