RPi-Distro/pi-gen #179: Wireless packages are not found in stage 2

spikespaz, 2018-04-07:
I get the following error, then the build exits.

```
Reading package lists... Done
Building dependency tree
Reading state information... Done
E: Unable to locate package wpasupplicant
E: Unable to locate package wireless-tools
```

I'm not sure what I need to do to add these to the available packages.

I get the same error on Ubuntu Xenial and Artful. Should I try Debian 9?

dbaba, 2018-04-07:
I got the same error as well. Tried `2017-11-29-raspbian-stretch` and `2018-03-13-raspbian-stretch` tags, but no luck. The scripts worked before, around mid-March. Were the package repositories broken?

dbaba, 2018-04-17:
@spikespaz I have a workaround. Create a script file containing the following code and save it to `stage2/01-sys-tweaks` dir as `02-run-chroot.sh`, and chmod 655 (+x) to it. Then the errors will be gone. Not sure it works for you, though.

```
apt-get update -y
```

dbaba, 2018-04-17:
Used `2018-03-13-raspbian-stretch` tag.

I can reproduce the errors by the following commands.
(I place SKIP files to build a lite image.)

```
git clone https://github.com/RPi-Distro/pi-gen.git
cd pi-gen/
git checkout -b 2018-03-13-raspbian-stretch
touch stage4/SKIP
touch stage5/SKIP
time ./build-docker.sh
```

Then I got,
```
        :
        :
[10:45:47] End /pi-gen/stage2/01-sys-tweaks/01-run.sh
[10:45:47] End /pi-gen/stage2/01-sys-tweaks
[10:45:47] Begin /pi-gen/stage2/02-net-tweaks
[10:45:47] Begin /pi-gen/stage2/02-net-tweaks/00-packages
Reading package lists...
Building dependency tree...
Reading state information...
E: Unable to locate package wpasupplicant
E: Unable to locate package wireless-tools

real	19m43.648s
user	0m0.708s
sys	0m0.276s
```

The `E: Unable to locate package ...` messages usually appear when the repository info isn't yet updated on performing `apt-get install` command. So I created the above chroot script in order to update the local repo prior to stage2 scripts.

The pi-gen scripts definitely worked before so I'd say these errors are brought by other stuff (but I cannot identify it).

XECDesign, 2018-04-17:
You need this commit: https://github.com/RPi-Distro/pi-gen/commit/93db3736721d968a6ab67d37243d6f3aa6958d00

dbaba, 2018-04-17:
Got it, thanks.

spikespaz, 2018-04-20:
@dbaba If that suggestion solved it for you I'll take your word and close this issue. I can't test it.

XECDesign, 2018-05-04:
No activity

closed 2018-05-04 by XECDesign