RPi-Distro/pi-gen #181: Is there a good way to add an out-of-tree kernel module to pi-gen?

cygnusnow, 2018-04-24:
Since the pi-gen doesn't compile the kernel from source, if I have a out-of-tree kernel module need to be added to the image, is there a good way to add to pi-gen?

XECDesign, 2018-04-24:
Do you want the modules to be rebuilt each time the kernel is updated or are you okay with it only working until the kernel is updated?

cygnusnow, 2018-04-26:
Since we have modified the kernel source, we hope the modified kernel is built when we try to generate a new Raspbian image. And there are several other modules are out-of-tree kernel module, we don't want to merge into the kernel source. So we also want to find a good way to rebuild the out-of-tree kernel module each time. But I notice that the dist is upgraded in export-image/01-set-sources/01-run.sh, so it is possible that if the Raspbian kernel is updated, the out-of-tree kernel module may not work any more because of the kernel version is mismatch.

I add a patch in stage4 to build our local kernel, but the  export-image/01-set-sources/01-run.sh may update the kernel before generate the new image.  It is not what we want.

XECDesign, 2018-05-04:
I'd install dkms, raspberrypi-kernel-headers and create a dkms package for your module. Then any time the kernel is updated, the module should be rebuilt automatically.

XECDesign, 2018-05-04:
Closing since it's not a pi-gen issue.

closed 2018-05-04 by XECDesign

XECDesign, 2018-08-30:
The same options apply as when cross-compiling the kernel itself. ARCH=arm CROSS_COMPILE=...