RPi-Distro/pi-gen #183: Docker Stage 0 fail: debootstrap directory not empty.

moriczgergo, 2018-04-29:
I'm trying to build my own custom image.

It seems that a debootstrap.log remains in the debootstrap dir, and rmdir can't remove the directory because of that.

If I remove it, it fails on /etc/apt not existing.

```
update-binfmts: warning: qemu-aarch64 already enabled in kernel.
[10:10:07] Begin /pi-gen
[10:10:07] Begin /pi-gen/stage0
[10:10:07] Begin /pi-gen/stage0/prerun.sh
I: Running command: debootstrap --arch armhf --foreign --components=main,contrib,non-free --keyring /pi-gen/stage0/files/raspberrypi.gpg stretch /pi-gen/work/2018-04-29-Fusee Framboise/stage0/rootfs http://raspbian.raspberrypi.org/raspbian/
I: Retrieving InRelease
I: Checking Release signature
I: Valid Release signature (key id A0DA38D0D76E8B5D638872819165938D90FDDD2E)
rmdir: failed to remove '/pi-gen/work/2018-04-29-Fusee Framboise/stage0/rootfs/debootstrap': Directory not empty
```

hongquan, 2018-09-15:
I don't have space in IMG_NAME, but still encounter this "debootstrap directory not empty" error.

The `rmdir` command was used for deleting the directory, but my directory has 3 files inside, so `rmdir` refused to do its job.

OneStromberg, 2019-10-01:
@hongquan so how the issue was solved?

hongquan, 2019-10-02:
@OneStromberg I no longer use Docker build. It seems to be that the build script used wrong version of debootstrap. I already go back to traditional method (not through Docker).