RPi-Distro/pi-gen #191: Lockfile issue after stage 3

Babbleshack, 2018-06-07:
`CONTINUE=1 PRESERVE_CONTAINER=1 ./docker-build.sh` has started failing on the package stage, with error:
```
[21:00:01] Begin /pi-gen/stage4/00-enable-ssh/00-packages
E: Could not open lock file /var/lib/dpkg/lock - open (2: No such file or directory)                               
E: Unable to lock the administration directory (/var/lib/dpkg/), are you root?  
```

I have removed the export files from `stage3`

XECDesign, 2018-06-08:
I think that would only happen if you're running two instanced of pi-gen at the same time, using the same files, or if you are continuing a build which was killed while dpkg was running.

closed 2018-06-12 by XECDesign

Babbleshack, 2018-06-12:
That doesnt seem to be the case. This will happen even when continuing a successful build.

XECDesign, 2018-06-13:
In that case, could you make sure everything is unmounted, then build the full stock image from scratch and post the log (something like `sudo ./build.sh |& tee build.log` would work)?