RPi-Distro/pi-gen #198: Building in VirtuallBox, Docker or Travis CI

filips123, 2018-07-14:
Is it possible to build image in VirtualBox, Docker or Travis CI?

XECDesign, 2018-07-14:
Yes to all of the above. There are a few gotchas, in that you need to be able to mount filesystems and have a certain module loaded, which could be a problem if you're running in an unpriviledged container without appopriate permissions. There's a script included for dockeer builds (see the README).

However, there's no support if something goes wrong.

filips123, 2018-07-15:
How long it will take to build image on those systems? Will be longer than on normal system?

Also, is building on Linux subsystem on Windows 10 (bash for ubuntu for windows 10) supported?

XECDesign, 2018-07-15:
There shouldn't be much overhead from virtualisation or running in a container.

> Also, is building on Linux subsystem on Windows 10 (bash for ubuntu for windows 10) supported?

No

filips123, 2018-07-15:
So, will building on Ubuntu 18.04 in VirtualBox on Windows 10 work?
What else can I do if I don't have computer with Linux?

closed 2018-07-24 by filips123