RPi-Distro/pi-gen #209: The following NEW packages will be installed:   libc-l10n locales : HANGS NOTHING FURTHER

outbackdingo, 2018-09-23:
appears to be hng here for hhours? any ideas?

Processing triggers for libc-bin (2.24-11+deb9u3) ...
Processing triggers for initramfs-tools (0.130) ...
[11:44:38] End /pi-gen/stage0/00-configure-apt/00-run.sh
[11:44:38] End /pi-gen/stage0/00-configure-apt
[11:44:38] Begin /pi-gen/stage0/01-locale
[11:44:38] Begin /pi-gen/stage0/01-locale/00-debconf
[11:44:39] End /pi-gen/stage0/01-locale/00-debconf
[11:44:39] Begin /pi-gen/stage0/01-locale/00-packages
Reading package lists...
Building dependency tree...
Reading state information...
The following additional packages will be installed:
  libc-l10n
The following NEW packages will be installed:
  libc-l10n locales

outbackdingo, 2018-09-24:
pstree -a                                                                                                                                                                                                                                          15:13  24.09.18  49.18G RAM 
systemd --switched-root --system --deserialize 32
  ├─ModemManager
  │   └─2*[{ModemManager}]
  ├─NetworkManager --no-daemon
  │   ├─dhclient -d -q -sf /usr/libexec/nm-dhcp-helper -pf /var/run/dhclient-enp2s0f1.pid -lf /var/lib/NetworkManager/dhclient-2488fb87-cdf7-3b1d-b78d-b1ecff166bc8-enp2s0f1.lease -cf /var/lib/NetworkManager/dhclient-enp2s0f1.conf enp2s0f1
  │   └─2*[{NetworkManager}]
  ├─abrt-applet
  │   └─3*[{abrt-applet}]
  ├─abrt-dbus -t133
  │   └─2*[{abrt-dbus}]
  ├─abrt-dump-journ -fxtD
  ├─abrt-dump-journ -fxtD
  ├─abrt-dump-journ -D -T -f -e
  ├─abrtd -d -s
  │   └─2*[{abrtd}]
  ├─accounts-daemon
  │   └─2*[{accounts-daemon}]
  ├─agent
  │   └─2*[{agent}]
  ├─akonadi_control
  │   ├─akonadi_akonote --identifier akonadi_akonotes_resource_0
  │   │   └─3*[{akonadi_akonote}]
  │   ├─akonadi_archive --identifier akonadi_archivemail_agent
  │   │   └─4*[{akonadi_archive}]
  │   ├─akonadi_birthda --identifier akonadi_birthdays_resource
  │   │   └─3*[{akonadi_birthda}]
  │   ├─akonadi_contact --identifier akonadi_contacts_resource_0
  │   │   └─3*[{akonadi_contact}]
  │   ├─akonadi_followu --identifier akonadi_followupreminder_agent
  │   │   └─4*[{akonadi_followu}]
  │   ├─akonadi_ical_re --identifier akonadi_ical_resource_0
  │   │   └─4*[{akonadi_ical_re}]
  │   ├─akonadi_indexin --identifier akonadi_indexing_agent
  │   │   └─3*[{akonadi_indexin}]
  │   ├─akonadi_maildir --identifier akonadi_maildir_resource_0
  │   │   └─3*[{akonadi_maildir}]
  │   ├─akonadi_maildis --identifier akonadi_maildispatcher_agent
  │   │   └─4*[{akonadi_maildis}]
  │   ├─akonadi_mailfil --identifier akonadi_mailfilter_agent
  │   │   └─4*[{akonadi_mailfil}]
  │   ├─akonadi_migrati --identifier akonadi_migration_agent
  │   │   └─3*[{akonadi_migrati}]
  │   ├─akonadi_newmail --identifier akonadi_newmailnotifier_agent
  │   │   └─3*[{akonadi_newmail}]
  │   ├─akonadi_sendlat --identifier akonadi_sendlater_agent
  │   │   └─4*[{akonadi_sendlat}]
  │   ├─akonadiserver
  │   │   ├─mysqld --defaults-file=/home/dingo/.local/share/akonadi/mysql.conf --datadir=/home/dingo/.local/share/akonadi/db_data/ --socket=/tmp/akonadi-dingo.CmMJ8s/mysql.socket --pid-file=/tmp/akonadi-dingo.CmMJ8s/mysql.pid
  │   │   │   └─33*[{mysqld}]
  │   │   └─25*[{akonadiserver}]
  │   └─2*[{akonadi_control}]
  ├─alsactl -s -n 19 -c -E ALSA_CONFIG_PATH=/etc/alsa/alsactl.conf --initfile=/lib/alsa/init/00main rdaemon
  ├─atd -f
  ├─auditd
  │   └─{auditd}
  ├─avahi-daemon
  │   └─avahi-daemon
  ├─baloo_file
  │   └─{baloo_file}
  ├─bluetoothd
  ├─chronyd
  ├─colord
  │   └─2*[{colord}]
  ├─crond -n
  ├─cupsd -l
  ├─dbus-daemon --system --address=systemd: --nofork --nopidfile --systemd-activation --syslog-only
  ├─dockerd
  │   ├─docker-containe --config /var/run/docker/containerd/containerd.toml
  │   │   ├─docker-containe -namespace moby -workdir /var/lib/docker/containerd/daemon/io.containerd.runtime.v1.linux/moby/527149810dac434cb9041ea7b5cae6e1e89ac548b2cccd5cf6caf06f878ddb18 -address /var/run/docker/containerd/docker-containerd.sock -containerd-binary /usr/bin/docker-containerd -runtime-root...
[... 338 lines cut ...]
  │   ├─mission-control
  │   │   └─3*[{mission-control}]
  │   ├─pulseaudio --daemonize=no
  │   │   ├─gconf-helper
  │   │   └─2*[{pulseaudio}]
  │   ├─xdg-desktop-por
  │   │   └─3*[{xdg-desktop-por}]
  │   ├─xdg-desktop-por
  │   │   └─10*[{xdg-desktop-por}]
  │   ├─xdg-desktop-por
  │   │   └─2*[{xdg-desktop-por}]
  │   ├─xdg-document-po
  │   │   └─5*[{xdg-document-po}]
  │   └─xdg-permission-
  │       └─2*[{xdg-permission-}]
  ├─systemd-journal
  ├─systemd-logind
  ├─systemd-udevd
  ├─udisksd
  │   └─4*[{udisksd}]
  ├─upowerd
  │   └─2*[{upowerd}]
  ├─wpa_supplicant -c /etc/wpa_supplicant/wpa_supplicant.conf -u -s
  └─xembedsniproxy
      └─2*[{xembedsniproxy}]

XECDesign, 2018-09-24:
By the looks of it, the chroot enviornment is having trouble with network access. Feels like a local issue.

outbackdingo, 2018-09-24:
is there something special about chrooting an app needing networking... im running on fedora and the only thing i can think of is a resolve.connf file missing from the environment however i have one.... 
and the host system networking is fully functional, and i can resolve domains and surf the web etc fine

cat /etc/resolv.conf                                                                                                                                                                                                                               
# Generated by NetworkManager
search lan
nameserver 192.168.10.1
nameserver fdcf:aa8a:cfb6::1

XECDesign, 2018-09-24:
Not that I'm aware of. Is that /etc/resolv.conf from your pi-gen work directory or the host system?

outbackdingo, 2018-09-24:
thats from the host

Ive rebooted now and shutdown all docker containers, ill give it one more shot, but i know its not going to work

outbackdingo, 2018-09-24:
from inside the container itself i can ping google this isnt a networking issue if you ask me

root@d6aaf4fbff09:/# ping www.google.com
PING www.google.com (172.217.161.164) 56(84) bytes of data.
64 bytes from hkg07s29-in-f4.1e100.net (172.217.161.164): icmp_seq=1 ttl=51 time=53.2 ms
^C
--- www.google.com ping statistics ---
1 packets transmitted, 1 received, 0% packet loss, time 0ms
rtt min/avg/max/mdev = 53.243/53.243/53.243/0.000 ms
root@d6aaf4fbff09:/#

outbackdingo, 2018-09-24:
root@d6aaf4fbff09:/# ping www.google.com
PING www.google.com (172.217.161.164) 56(84) bytes of data.
64 bytes from hkg07s29-in-f4.1e100.net (172.217.161.164): icmp_seq=1 ttl=51 time=53.2 ms
^C
--- www.google.com ping statistics ---
1 packets transmitted, 1 received, 0% packet loss, time 0ms
rtt min/avg/max/mdev = 53.243/53.243/53.243/0.000 ms
root@d6aaf4fbff09:/# ps xa
  PID TTY      STAT   TIME COMMAND
    1 ?        Ss     0:00 bash -e -o pipefail -c dpkg-reconfigure qemu-user-static && .cd /pi-gen; ./build.sh; .rsync -av work/*/build.log deploy/
   39 ?        S      0:00 /bin/bash -e ./build.sh
  665 ?        Sl     0:00 /usr/bin/qemu-arm-static /bin/bash
  667 ?        Sl     0:01 /usr/bin/qemu-arm-static /usr/bin/apt-get install -y locales
  679 ?        Sl     0:00 /usr/bin/qemu-arm-static /usr/lib/apt/methods/http
  682 ?        Sl     0:00 /usr/bin/qemu-arm-static /usr/lib/apt/methods/http
  685 pts/0    Ss     0:00 bash
  693 pts/0    R+     0:00 ps xa
root@d6aaf4fbff09:/# pftree
bash: pftree: command not found
root@d6aaf4fbff09:/# pstree
bash---build.sh---bash-+-apt-get-+-2*[http---{http}]
                       |         `-{apt-get}
                       `-{bash}
root@d6aaf4fbff09:/# apt-get update
Ign:1 http://deb.debian.org/debian stretch InRelease
Get:2 http://security.debian.org/debian-security stretch/updates InRelease [94.3 kB]
Get:3 http://deb.debian.org/debian stretch-updates InRelease [91.0 kB]
Get:4 http://deb.debian.org/debian stretch Release [118 kB]
Get:5 http://security.debian.org/debian-security stretch/updates/main amd64 Packages [441 kB]
Get:6 http://deb.debian.org/debian stretch-updates/main amd64 Packages [5148 B]
Get:7 http://deb.debian.org/debian stretch Release.gpg [2434 B]
Get:8 http://deb.debian.org/debian stretch/main amd64 Packages [7099 kB]
39% [8 Packages 1335 kB/7099 kB 19%]                                                                                                                                                  340 kB/s43% [8 Packages 1665 kB/7099 kB 23%]                                                                                                                                                  340 kB/s47% [8 Packages 2039 kB/7099 kB 29%]                                                                                                                                                  340 kB/s51% [8 Packages 2467 kB/7099 kB 35%]                                                                                                                                                  340 kB/s55% [8 Packages 2828 kB/7099 kB 40%]                                                                                                                                                  340 kB/s58% [8 Packages 3126 kB/7099 kB 44%]                                                                                                                                                  340 kB/s62% [8 Packages 3524 kB/7099 kB 50%]                                                                                                                                                  340 kB/s66% [8 Packages 3954 kB/7099 kB 56%]                                                                                                                                                   340 kB/71% [8 Packages 4411 kB/7099 kB 62%]                                                                                                                                                   340 kB/76% [8 Packages 4874 kB/7099 kB 69%]                                                                                                                                                   340 kB/80% [8 Packages 5334 kB/7099 kB 75%]                                                                                                                                                   340 kB/85% [8 Packages 5790 kB/7099 kB 82%]                                                                                                                                                   340 kB/90% [8 Packages 6248 kB/7099 kB 88%]                                                                                                                                                   818 kB/94% [8 Packages 6700 kB/7099 kB 94%]                                                                                                                                                   818 kB/98% [Working]                                                                                                                                                                          818 kB/98% [8 Packages store 0 B]                                                                                                                                                             818 kB/98% [8 Packages store 0 B]                                                                                                                                                             818 kB/100% [Working]                                                                                                                                                                         818 kB/Fetched 7851 kB in 13s (564 kB/s)
Reading package lists... Done
root@d6aaf4fbff09:/#

XECDesign, 2018-09-24:
The docker container and the chroot environment within the docker container that apt runs from are two different things.

outbackdingo, 2018-09-24:
ok lets go deeper chroot into make sure im correct, this still has networking
and......
root@d6aaf4fbff09:/# chroot /pi-gen/work/2018-09-24-my_name/stage0/rootfs/
# ping 8.8.8.8
PING 8.8.8.8 (8.8.8.8) 56(84) bytes of data.
Unsupported ioctl: cmd=0x8906
64 bytes from 8.8.8.8: icmp_seq=1 ttl=120 time=22.0 ms
^C
--- 8.8.8.8 ping statistics ---
1 packets transmitted, 1 received, 0% packet loss, time 0ms
rtt min/avg/max/mdev = 22.015/22.015/22.015/0.000 ms
# cat /etc/resolv.conf
# Generated by NetworkManager
search lan
nameserver 192.168.10.1
# cat /etc/resolv.conf
# Generated by NetworkManager
search lan
nameserver 192.168.10.1
# uname -a
Linux d6aaf4fbff09 4.18.9-200.fc28.x86_64 #1 SMP Thu Sep 20 02:43:23 UTC 2018 armv7l GNU/Linux
# apt-get update
Hit:1 http://archive.raspberrypi.org/debian stretch InRelease
Hit:2 http://raspbian.raspberrypi.org/raspbian stretch InRelease
Reading package lists... Done

outbackdingo, 2018-09-24:
So... from in the container i chrooted as shown tested for resolution and networking
which appear ok, i then tried 

root@d6aaf4fbff09:/# /usr/bin/apt-get install -y locales
Reading package lists... Done
Building dependency tree
Reading state information... Done
The following additional packages will be installed:
  libc-l10n
The following NEW packages will be installed:
  libc-l10n locales
0 upgraded, 2 newly installed, 0 to remove and 0 not upgraded.
Need to get 4107 kB of archives.
After this operation, 13.8 MB of additional disk space will be used.
0% [Working]

AND IT HANGS.....

XECDesign, 2018-09-24:
Absolutely no idea. Haven't seen that one before. =/

outbackdingo, 2018-09-24:
would it be poossible to provide the full path to the files in need i will try to download them and install manually from inside the container

XECDesign, 2018-09-24:
`apt --print-uris install package_name(s)`

outbackdingo, 2018-09-24:
Okay seems manually installing them and restarting the whole CONTINUE=1 ,/build.shh has moved me forward to stage 2

closed 2018-09-24 by outbackdingo