RPi-Distro/pi-gen #216: Add custom kernel module

csabakeszegh, 2018-11-01:
Hi,
For my project I would like to create a custom image (based on the -lite image), which has kernel module for PEAK USB CAN devices.  The upstream kernel has [peak_usb](https://github.com/raspberrypi/linux/tree/rpi-4.14.y/drivers/net/can/usb/peak_usb) module, and PEAK also provides a linux kernel [driver](https://www.peak-system.com/fileadmin/media/linux/index.htm).  Unfortunately `peak_usb` is not installed with the `raspberrypi-kernel` package.  On desktop Ubuntu there is a `linux-image-extra-virtual` which provides this kernel module, but could not find any similar for Raspbian.

My question is what is the recommended way to have additional kernel module in the image.
Thanks in advance,
csaba

csabakeszegh, 2018-11-01:
Thanks for the hint.  I'm afraid official `peak-linux-driver` is not DKMS compatible by default. It has a custom makefile.  However I found a repo that adds DKMS support: https://salsa.debian.org/debian/peak-linux-driver/  Is it possible to use this repo directly from the image build script, or should I first create a .deb package, upload to a ppa, and install from there during the build?

XECDesign, 2018-11-01:
I'd say it all depends on your preferred workflow. It's going to be messy either way.

If you use that git repo, you're still going to need to build the .deb from it and install it either directly through dpkg or an apt repo.

csabakeszegh, 2018-11-01:
I found a [post](https://www.peak-system.com/forum/viewtopic.php?t=2754), that walks through how to register `peak-linux-driver` with DKMS (without creating a package)  I think it might work for me too.  Now during build I install the source in `${ROOTFS_DIR}/usr/src/peak-linux-driver-8.6.0`, copy dkms.conf, and run the `dkms add` command in chroot.  After this I can run `dkms build/install` on the device. Currently if I try to run `dkms build` in chroot it will pick the host's kernel version and fail.

It would be great if I could run dkms during build, or on the first boot automatically.

csabakeszegh, 2018-11-02:
Finally, I managed to make it work. After installing `raspberry-kernel-headers` I extract the kernel versions from `/usr/src/linux-headers-<VERSION>` directory paths, and run `dkms build/install peak-linux-driver/8.6.0 -k <VERSION>` in chroot. So the modules will be available already on the first boot, and DKMS will run on kernel updates.  Thanks for pointing me in the right direction.

XECDesign, 2018-11-19:
Glad you got it working. Okay to close?

csabakeszegh, 2018-11-20:
Thank you for your help, closing the issue.

closed 2018-11-20 by csabakeszegh