RPi-Distro/pi-gen #217: Failures inside on_chroot do not stop the build

tomas-c, 2018-11-06:
Most build scripts have "-e" option which is meant to make failed commands interrupt the script.

However, if any commands fail inside `on_chroot` call, the build still continues.

This seems a bit risky as you can only notice failures by carefully analysing build logs.

Replacing all instances of `on_chroot` with `on_chroot -e` fixes this.

XECDesign, 2018-11-06:
Good idea. Does only changing https://github.com/RPi-Distro/pi-gen/blob/master/scripts/common#L91 to `capsh --drop=cap_setfcap "--chroot=${ROOTFS_DIR}/" -- "-e $@"` do the job?

tomas-c, 2018-11-08:
It works if I move -e outside the quotes. I have created #219 with the necessary code changes.

PR #219 (2018-11-08): scripts: do not ignore errors inside on_chroot calls
Fixes #217
closed 2018-11-08 by XECDesign