RPi-Distro/pi-gen #222: init_1.48_armhf.deb install problem

zoff99, 2018-11-22:
```
dpkg: version 0.8.19 of ifupdown already installed, skipping
dpkg: version 6.3.0.dfsg.1-1+b1 of info already installed, skipping
dpkg: regarding .../archives/init_1.48_armhf.deb containing init, pre-dependency problem:
 init pre-depends on systemd-sysv | sysvinit-core
  systemd-sysv is unpacked, but has never been configured.
  sysvinit-core is not installed.

dpkg: error processing archive /var/cache/apt/archives/init_1.48_armhf.deb (--unpack):
 pre-dependency problem - not installing init
```

how can i fix this

zoff99, 2018-11-22:
this seems to be related:
https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=857338

zoff99, 2018-11-22:
its with ubuntu 14.04 on circleCI
are you guys building this on an CI system?

XECDesign, 2018-11-22:
No, pi-gen is currently just running on a nightly cronjob. There's a CI system for the kernel and firmware, but images are tested manually before release.

Are you building from a jessie-based system/environment?

zoff99, 2018-11-23:
now i am building from ubuntu 14.04 then docker inside that. seems to work now.
takes about 30 minutes on CI

XECDesign, 2018-11-23:
Okay to close?

hariavers, 2019-04-14:
Still failing for stage 0 build @ 14/04/2019 build generated by commit e910bf55ac92e27c6e77043e8995bcb5e0a2d742

The erro is in work/2019-04-14-Raspbian-hari/stage0/rootfs/debootstrap/debootstrap.log

dpkg: version 2:8.0.0197-4+deb9u1 of xxd already installed, skipping
Errors were encountered while processing:
 /var/cache/apt/archives/init_1.48_armhf.deb

How can this be fixed?
[debootstrap.log](https://github.com/RPi-Distro/pi-gen/files/3077402/debootstrap.log)