RPi-Distro/pi-gen #223: Bad uname -r kernel version

Yes21, 2018-12-05:
We thank you very much for this nice and very useful script.
We have modified it to build our specific version of raspbian for our project ([EZ-Wifibroadcast](https://github.com/RespawnDespair/wifibroadcast-image-builder)).

We encounter a problem that we can't solve : 
When we are in CHROOT stages, the `uname -r` command returns the kernel number of the host, but not those of the raspbian we have installed !

Did you ever had this problem, and do you know how to solve it ?

Many thanks for help.

XECDesign, 2018-12-05:
Since uname prints system information for the currently running kernel, that's working as intended.

Here's how we're working around that:
https://github.com/RPi-Distro/pi-gen/blob/master/export-image/04-finalise/01-run.sh#L61

I'm sure you could also extract the string from the actual kernel image.

Yes21, 2018-12-05:
Thank you very much for your fast answer.
I will look at this ...

Yes21, 2018-12-07:
Many thanks to have pointing us into the right direction.
We could solve our problem !

closed 2018-12-07 by Yes21

XECDesign, 2018-12-07:
Glad you found a workaround