RPi-Distro/pi-gen #225: Idea: Factorize image filename generation

hongquan, 2018-12-10:
My project (OSS) need to build a custom Raspbian with custom filename (include more info, like Git revision, Git branch). I find that, to do that, I have to modify 3 places of `pi-gen`. I think that it is not so good. Do you think that `pi-gen` should have a single place to define filename?

XECDesign, 2018-12-11:
If you have a clean way of doing it that keeps everything the same for Raspbian filenames by default, it sounds like a good idea.

bjsowa, 2018-12-21:
I fixed it with #227 
Now you can export `IMG_FILENAME` for image file and `ZIP_FILENAME` for deployment zip file

bjsowa, 2019-01-04:
@AlexSwensen the exported image should be in:
`${WORK_DIR}/${IMG_DATE}-${IMG_NAME}/export-image/${IMG_FILENAME}.img`
and the deployment zip file in:
`${DEPLOY_DIR}/${ZIP_FILENAME}.zip`

IMG_FILENAME defaults to:
`${IMG_DATE}-${IMG_NAME}${IMG_SUFFIX}`
and ZIP_FILENAME to:
`image_${IMG_DATE}-${IMG_NAME}${IMG_SUFFIX}`

I didn't change the paths these files are generated or their default filenames. I only added an option to specify custom filenames.