RPi-Distro/pi-gen #228: Idea: Add custom first user to passworless sudo like 'pi' user

moimart1, 2018-12-21:
Now it's possible to create a custom username/password with `FIRST_USER_*` variables but the custom user is not passwordless like `pi` user because `pi` user uses `/etc/sudoers.d/010_pi-nopasswd`

It is possible to patch the sudoers file to remove `pi` user and add the custom user ?

Maybe, somethings like that:
```
if [ ! -f "${ROOTFS_DIR}/etc/sudoers.d/010_${FIRST_USER_NAME}-nopasswd" ]; then
    rm -f "${ROOTFS_DIR}/etc/sudoers.d/010_pi-nopasswd"
    echo "${FIRST_USER_NAME} ALL=(ALL) NOPASSWD: ALL" > "${ROOTFS_DIR}/etc/sudoers.d/010_${FIRST_USER_NAME}-nopasswd"
    chmod 0440 "${ROOTFS_DIR}/etc/sudoers.d/010_${FIRST_USER_NAME}-nopasswd"
fi
```

samtygier, 2019-01-06:
I missed that when searching for all the places that the username was specified. It seems that it is hardwired into the raspberrypi-sys-mods package. I think it is bad form for the scripts to modify a file from a deb package, as it will be overwritten on updates.

Would a better fix be to ask the raspberrypi-sys-mods developers to apply NOPASSWD to the `sudo` group instead of the `pi` user?

XECDesign, 2019-01-07:
I think it was a mistake to add the nopasswd file to sys-mods. I was moving changes out of pi-gen and into packages where appropriate and got a bit overzealous.

There are few ways to go here.
1) Make the nopasswd file the responsibility or pi-gen again and take it out of sys-mods
2) Add a debconf option to sys-mods which would allow us to pre-seed the nopasswd use
3) As you say, have pi-gen override sys-mods

2 was the original plan and would allow the user to be asked what they would like to do. There's understandable concern of having a grant root access without prompting the user in any way.

rkubes, 2019-01-15:
I would vote for option 2. I'm one of the people that don't like the idea of NOPASSWD, and would prefer the option to disable it if possible. I currently work around this by having pi-gen delete the pi-nopasswd file in stage2. Though, to samtygier's point, if it ever does come back in an update, I would be unaware.

XECDesign, 2019-01-16:
Regarding updates, files in /etc are special. Since they're config files, it's expected that the system admin will modify them and there's nothing wrong with doing so. Debian will err on the side of caution and ask you what you want to do, if the system admin and the package maintainer both try to make changes.

In this case, we'd probably want to avoid that situation since the distro is aimed at beginners, most of whom would have no idea what they're being asked. Having the package handle the file in a beginner-friendly, safe and flexible manner starts getting squirrely. That's part of the reason I'm leaning toward 1. I'd like to avoid any potential snafus when it comes to handling root privileges.

mikysal78, 2019-05-01:
i have this:
sudo: /usr/bin/sudo must be owned by uid 0 and have the setuid bit set

mikysal78, 2019-05-04:
`if [ ! -f "${ROOTFS_DIR}/etc/sudoers.d/010_${FIRST_USER_NAME}-nopasswd" ]; then
    rm -f "${ROOTFS_DIR}/etc/sudoers.d/010_pi-nopasswd"
    echo "${FIRST_USER_NAME} ALL=(ALL) NOPASSWD: ALL" > "${ROOTFS_DIR}/etc/sudoers.d/010_${FIRST_USER_NAME}-nopasswd"
    chmod 0440 "${ROOTFS_DIR}/etc/sudoers.d/010_${FIRST_USER_NAME}-nopasswd"
    chown -R root:root /usr
    chmod 4755 /usr/bin/sudo
fi`

fuji246, 2021-09-15:
> i have this:
> sudo: /usr/bin/sudo must be owned by uid 0 and have the setuid bit set

@mikysal78 , I have the same problem, wondering what is causing the problem here?