RPi-Distro/pi-gen #232: move  init_resize.sh to /boot allows user choice for resize

DG12, 2019-01-02:
Users that need anything other than having the root partition and filesystem to occupy the entire SD card have a very difficult time. The  init_resize.sh script is located on a partition that cannot be modified by windows or Mac OS. If it were located on the FAT /boot partition it could be easily modified.
This change is simple and opens many options for the initial booting.

1) Modify the script to use the new location:
change:
`sed -i 's| init=/usr/lib/raspi-config/init_resize.sh||' /boot/cmdline.txt`
to:
`sed --in-place=o 's| init=.*sh||' /boot/cmdline.txt  # once resized do not invoke me again. `

2) move init_resize.sh to /boot

This will NOT change existing operations and requires no additional maintenance.

Thank you

XECDesign, 2019-03-06:
Although the resize script is on a partition that can't be reached from windows, the user can still take out the init= parameter from cmdline.txt if they choose, or provide their own resize script.

closed 2019-03-06 by XECDesign

DG12, 2019-03-06:
Would it really be that hard to make the change I suggested?

Disappointed

mickuehl, 2020-04-02:
resizing is just one task that can run on a first boot, I would like a generic option like proposed above to allow customisation while booting the first time. Yes I know it can be done from a Linux system but it would be nice to have a simple option from Windows and MacOS.

CCCPart, 2021-02-28:
error ( -0) just changed to (-117)

mangelozzi, 2021-11-28:
Why was this issue closed?

mangelozzi, 2021-11-28:
> Although the resize script is on a partition that can't be reached from windows, the user can still take out the init= parameter from cmdline.txt if they choose, or provide their own resize script.

Could you please provide an example of how to make the raspberry pi OS resize to say 4GB?

DG12, 2021-11-28:
BEFORE booting the image for the First Time ,
Take out the` init= `parameter from /`boot/cmdline.txt` this will prevent the resize script from running on first boot.

If you will be running headless (i.e. no keyboard, no monitor only access via ssh) create a file named `/boot/ssh `with any (or no) contents.

Then if you will be running from a wifi network create 
`/boot/wpa-supplicant.conf` which the boot process will use to overwrite `/etc/wpa_supplicant/wpa-supplicant.conf` and then deletes the boot one.
For example:

```
ctrl_interface=DIR=/var/run/wpa_supplicant GROUP=netdev
update_config=1
country=US

network={
    ssid="my-netname"
    psk="Myi-wifi-NetPass"
}

```
Check for matching quotes

Boot the image.
Then use `sudo raspi-config` ; advanced option to expand the file system.

You might consider leaving enough room for another volume like. `/varnew `or `/data`