RPi-Distro/pi-gen #239: Unable to locate package apt-transport-https

bzhmaddog, 2019-01-16:
In stage0/00-configure-apt/00-run.sh I am adding a new packages sources

install -m 644 files/snips.list '${ROOTFS_DIR}/etc/apt/sources.list.d/'
on_chroot apt-key add - < files/snips.gpg.key
Theses two steps works,
But later in when doing apt-get update the update fail when trying to retrieve packages list from snips

Reading package lists ... Done
E: The method driver /usr/lib/apt/methods/https could not be found
N: Is the package apt-transport-https installed?
E: Failed to fetch https:://raspbian.snips.ai/stretch/dists/stable/InRelease
E: Some index files failed to download. They have been ignored, or old ones used instead.

I tried to add the package just before the update is done but it doesn't work all the time

XECDesign, 2019-01-18:
Do you have a public branch somewhere or a diff with your changes?

bzhmaddog, 2019-01-23:
Nope. It is an internal project for my company I can't publish anything public yet

XECDesign, 2019-01-23:
A minimal testcase?

XECDesign, 2019-01-24:
Some things that jump out
1) `https:://` not sure if that's a type in the sources file or just how apt formats the output
2) If 1 isn't the problem, have you 100% verified that apt-transport-https  is actually installed first?

bzhmaddog, 2019-01-24:
I managed to resolve my issue. I changed re-ordering my commands

closed 2019-01-24 by XECDesign

bzhmaddog, 2019-01-24:
Here is the content of my 'stage0/00-configure-apt/00-run.sh'

#!/bin/bash -e

install -m 644 files/sources.list "${ROOTFS_DIR}/etc/apt/"
install -m 644 files/raspi.list "${ROOTFS_DIR}/etc/apt/sources.list.d/"

if [ -n "$APT_PROXY" ]; then
	install -m 644 files/51cache "${ROOTFS_DIR}/etc/apt/apt.conf.d/51cache"
	sed "${ROOTFS_DIR}/etc/apt/apt.conf.d/51cache" -i -e "s|APT_PROXY|${APT_PROXY}|"
else
	rm -f "${ROOTFS_DIR}/etc/apt/apt.conf.d/51cache"
fi

on_chroot apt-key add - < files/raspberrypi.gpg.key

on_chroot << EOF
apt-get update
apt-get dist-upgrade -y
apt-get install -y apt-transport-https
EOF

install -m 644 files/snips.list "${ROOTFS_DIR}/etc/apt/sources.list.d/"
on_chroot apt-key add - < files/snips.gpg.key

on_chroot << EOF
apt-get update
EOF