RPi-Distro/pi-gen #244: Unable to remove existing "work" directories

brunp, 2019-01-23:
It appears that none of the "work" directories can be deleted after completing a Raspbian build. When I attempt to remove dated directories (whether it be as root, using sudo, or normally), I am either presented with "invalid permissions" or "operation not permitted". Not being able to deleted previous work directory builds unnecessarily increases disk space.

XECDesign, 2019-01-23:
Since there's no proper error recovery in pi-gen, this could happen if there's an error during install and  sys/proc/dev or the image loopback devices are mounted.

If there are no open files or directories, I don't really have any idea why that would happen. I haven't run into this problem myself.

XECDesign, 2019-01-23:
`mount` and `losetup -a` may give hints. If you reboot the VM, can you remove the directories?

brunp, 2019-01-24:
yes, I can remove the directories when I reboot which leads me to believe there is some process or lock on a specific directory in the build.

brunp, 2019-01-24:
I will investigate and report back to see if it specific to my set up.

pinkynrg, 2019-04-22:
any news about this issue? Facing the same problem.

pinkynrg, 2019-04-25:
@XECDesign 

> Since there's no proper error recovery in pi-gen, this could happen if there's an error during install and sys/proc/dev or the image loopback devices are mounted.

What would be the proper way to unmount the image loopback devices?

I'm running this on CI and every time it fails the runner has to be restarted in order for it to be able to delete the work folder.

XECDesign, 2019-04-25:
Check the output of 'mount'. You may need to run something like `umount some_path/{sys,proc,dev} -Rv` If any loopback devices (/dev/loop*) are mounted `umount` them too. Run `losetup -a` to check if any image files are attached to any of the loopback devices. If there are, run `losetup -D /dev/loopN` for the appropriate devices. I don't know how docker affects this, so maybe you'll need to do this from inside the container, maybe not.

pinkynrg, 2019-04-30:
Fantastic thanks it worked using the following command in my CI:

```
# killing build.sh process if it is executing
- "sudo kill $(ps aux | grep '[b]uild.sh' | awk '{print $2}') || :"
# umounting partitions in case last build failed
- "sudo umount --recursive /opt/pi-gen/work/{stage1,stage0,stage2,stage-base,stage-sbb,stage-hmi}/rootfs/{dev,proc,sys} || :"
```

Taha-Firoz, 2023-05-17:
This issue is open since 2019, @pinkynrg can you confirm if this works. I accidentally wiped by ubuntu install but `rm -rf`ing my work folder