RPi-Distro/pi-gen #245: Speeding up build time by using "SKIP" works for the current day only

brunp, 2019-01-23:
It appears that at the start of a new day, you can not re-use the previously build "stages" to increase build times. If you are only making modifications in stage 2 or 3, how do you skip stage 0 and 1 builds on a regular basis. It seems that on each new day, a new directory is generated with the present day timestamp as the directory name. At the moment, my work around is to rename the directory, but that is not a solution.

brunp, 2019-01-24:
I have changed the work directory target in the build.sh script. I guess this is not something that the community would want (i.e., pushing it back)? I assume fixes and/or submissions are reviewed/voted?

XECDesign, 2019-01-24:
> I assume fixes and/or submissions are reviewed/voted?

No, to the dismay of some, I just exercise my own subjective opinion. I try to reign in feature creep and keep it simple.

In this case, using IMG_DATE for the work directory was just a bad design decision. If there's a nice PR that fixes this and is able to run multiple builds simultaneously, I'd accept it.

brunp, 2019-01-24:
I got to run a couple tests to ensure that changes get picked up, when changes are made without using SKIP. I think I noticed at one point that on the same day, running the build skipped the stage 0/1 but picked up modified files. I will investigate and confirm over the weekend.

brunp, 2019-01-24:
@XECDesign What do you mean by "nice PR"?

XECDesign, 2019-01-24:
A pull requests that fixes the problem and has no downsides

brunp, 2019-01-24:
alright, a nice clean pull.

FredericGuilbault, 2019-03-14:
Been able to set IMG_DATE in config to something other then date would be a good workaround. 
So `IMG_DATE="static"` would create `./work/static-raspbian/...`