RPi-Distro/pi-gen #246: Method to introduce a new file in /boot/ or /etc/ directory?

brunp, 2019-01-23:
This is likely a spin off of issue #127, but what is the proper method? I am introducing a new script that is called by rc.local. The script then returns rc.local to its original state after it is run, but the new file, although visible in the "staged" directories is not present in the final image. After looking at #127, someone mentioned the OP was copying to /rpicluster/ instead of the intended target? I had made an adjustment in the one of the scripts to copy the new file to its intended target, but when the image is generated, it seems to skip the file. 

Would appreciate tips?

XECDesign, 2019-01-24:
Could you make a minimal testcase patch that demonstrates the issue? Are you making sure to do a clean build? Are you using docker (if so, I can't help)?

brunp, 2019-01-24:
I will see if I can make a testcase patch, but this is definitely performed on a clean build and I am NOT using docker. 

Right now, I am copying the new file to the boot directory in stage2? Should I be placing the file in the boot directory in an earlier stage perhaps? I can see my file in the list of files copied during the build, but hard to tell what is copied into the exported image.

XECDesign, 2019-01-24:
It doesn't sound like you're doing anything wrong.

brunp, 2019-01-24:
At the moment, I modified 01-run.sh in stage2/01-sys-tweaks and added the following line under the install of rc.local:

```
install -m 755 files/rc.local           "${ROOTFS_DIR}/etc/"
install -m 755 files/rpi_init           "${ROOTFS_DIR}/boot/"
```
The file obviously exists in the files directory. The intent is as before is to have rc.local run rpi_init which runs a script and returns rc.local to is original file contents. 

Would be good to know if I am missing any other logic?

brunp, 2019-01-24:
that is correct. When I launch the pi device for the first time, rc.local complains that rpi_init is missing? When I extract the image before even launching the PI device and review the boot files, the file is missing.

Paul

XECDesign, 2019-01-24:
Is it in every stage after stage 1 in the work directory? `ls work/*/*/rootfs/boot/rpi_init`

brunp, 2019-01-24:
I can confirm it is in both stage2 and stage3 boot directory.

brunp, 2019-01-24:
How do I pass a testcase on to you in these situations?

XECDesign, 2019-01-24:
Up to you. Either fork the repo and add the changes to your own branch or attach a patch here.

brunp, 2019-01-24:
I can fork a branch in my account and push up the changes. Might be a little while by the time it is ready, but within an hour.

XECDesign, 2019-01-24:
No rush, I probably won't be able to take a look before tomorrow anyway.

brunp, 2019-01-24:
Just proved to myself that it works with my simple testcase, so more to investigate in my since in my other repo. The only difference I see is that I simplified the script to two echo statements that copy to a log file. So, having said that, I need to see what else is different with my other repo that is resulting in a missing file from the file build.

brunp, 2019-01-24:
Not sure what the issue was, but now, my other repo has the file in the boot directory. Perhaps I had a partial build, not sure. However, this is no longer an issue. I will close this up now.

closed 2019-01-24 by brunp