RPi-Distro/pi-gen #247: modprobe: ERROR: ../libkmod/libkmod.c:586 kmod_search_moddep() could not open moddep file

r4and0m0, 2019-01-28:
Hello, while building (sudo build.sh) i get this error :

Setting up libav-tools (7:3.2.12-1~deb9u1+rpt1) ...
Setting up python3-pgzero (1.2.post1-2) ...
Setting up python3-tk:armhf (3.5.3-1) ...
Setting up python-tk (2.7.13-1) ...
Setting up sense-hat (1.2) ...
Enabling I2C...
modprobe: ERROR: ../libkmod/libkmod.c:586 kmod_search_moddep() could not open moddep file '/lib/modules/4.15.0-43-generic/modules.dep.bin'
modprobe: FATAL: Module i2c-dev not found in directory /lib/modules/4.15.0-43-generic
Setting up python-requests-oauthlib (0.7.0-0.1) ...
Setting up idle-python2.7 (2.7.13-2+deb9u3) ...
Setting up alacarte (3.11.91-2+rpi5) ...
Setting up python3-picamera (1.13) ...
Setting up idle (2.7.13-2) ...

XECDesign, 2019-01-28:
That's fine. The sense-hat package is trying to load i2c-dev, but that's not going to work in the chroot environment. That's safe to ignore.

r4and0m0, 2019-02-01:
Tyvm for the explanation.

closed 2019-02-01 by r4and0m0