RPi-Distro/pi-gen #248: debootstrap not empty

jasonehines, 2019-01-28:
I just started getting this error today (was working fine yesterday): `rmdir: failed to remove '/root/pi-gen/work/2019-01-28-Raspbian/stage0/rootfs/debootstrap': Directory not empty`

The bottom of the debootstrap.log:
`/lib/ld-linux-armhf.so.3: No such file or directory`

This is the contents of `lib`: 
```
drwxr-xr-x  9 root root  4096 Dec 18  2016 .
drwxr-xr-x 17 root root  4096 Feb  4  2017 ..
drwxr-xr-x  3 root root  4096 Jan 28 17:29 arm-linux-gnueabihf
drwxr-xr-x  2 root root  4096 May 21  2014 init
-rwxr-xr-x  1 root root 76428 May 18  2016 klibc-z-GH4vbo1mvhrmAe8pZWxAgGeP0.so
lrwxrwxrwx  1 root root    30 Jan 14  2018 ld-linux-armhf.so.3 -> arm-linux-gnueabihf/ld-2.24.so
lrwxrwxrwx  1 root root    20 Nov 15  2016 libnih-dbus.so.1 -> libnih-dbus.so.1.0.0
-rw-r--r--  1 root root 34212 Nov 15  2016 libnih-dbus.so.1.0.0
lrwxrwxrwx  1 root root    15 Nov 15  2016 libnih.so.1 -> libnih.so.1.0.0
-rw-r--r--  1 root root 87472 Nov 15  2016 libnih.so.1.0.0
drwxr-xr-x  3 root root  4096 Dec 18  2016 lsb
drwxr-xr-x  2 root root  4096 Dec 26  2016 modprobe.d
drwxr-xr-x  3 root root  4096 Dec 18  2016 systemd
drwxr-xr-x 15 root root  4096 Dec 28  2017 terminfo
drwxr-xr-x  3 root root  4096 Mar 17  2017 udev
```
This is the output of `ls -la arm-linux-gnueabihf/ld-*`:
```
-rwxr-xr-x 1 root root 138576 Jan 14  2018 ld-2.24.so
lrwxrwxrwx 1 root root     10 Jan 14  2018 ld-linux-armhf.so.3 -> ld-2.24.so
lrwxrwxrwx 1 root root     19 Jan 14  2018 ld-linux.so.3 -> ld-linux-armhf.so.3
```
I tested this on a fresh clone of the repo with this in the config file `IMG_NAME="Raspbian"`

Any idea what is causing this?

battuashwik, 2019-02-06:
Any update on this, having the same issue, even on the last release commit.

battuashwik, 2019-02-08:
@jasonehines  I just got it working, seems like apt-sources are outdated in the apt-ng-cache, disable apt cache and rebuild from scratch, worked for me.

samtygier, 2019-02-17:
I also see this error with APT_PROXY miss-configured. Disabling or correcting APT_PROXY lets me build again.

XECDesign, 2019-03-11:
@jasonehines any progress?

jasonehines, 2019-03-12:
I ended up just wiping my virtual machine and starting fresh

XECDesign, 2019-03-12:
Alright, I'll close it for now then. Thanks.

closed 2019-03-12 by XECDesign