RPi-Distro/pi-gen #253: custom config with build-docker.sh

RafalSkolasinski, 2019-01-31:
When using script to build image in docker `build-docker.sh` the `-c` command line argument to specify different config file does not seem to work.

RafalSkolasinski, 2019-01-31:
More logs
```
machine42 ➜  pi-gen git:(master) ✗ ./build-docker.sh -c customconfig
Sending build context to Docker daemon  883.2kB
Step 1/5 : FROM debian:stretch
 ---> a0bd3e1c8f9e
Step 2/5 : ENV DEBIAN_FRONTEND noninteractive
 ---> Using cache
 ---> ddfe33ffef49
Step 3/5 : RUN apt-get -y update &&     apt-get -y install         git vim parted         quilt realpath qemu-user-static debootstrap zerofree pxz zip dosfstools         bsdtar libcap2-bin rsync grep udev xz-utils curl xxd file     && rm -rf /var/lib/apt/lists/*
 ---> Using cache
 ---> 2a6986b03e65
Step 4/5 : COPY . /pi-gen/
 ---> fa8d6ab31d85
Step 5/5 : VOLUME [ "/pi-gen/work", "/pi-gen/deploy"]
 ---> Running in 0ef51cfb3594
Removing intermediate container 0ef51cfb3594
 ---> e0d17ecf358d
Successfully built e0d17ecf358d
Successfully tagged pi-gen:latest
update-binfmts: warning: qemu-aarch64 already enabled in kernel.
[20:11:35] Begin /pi-gen
[20:11:35] Begin /pi-gen/'stage0
./build.sh: line 85: pushd: /pi-gen/'stage0: No such file or directory
```

whereas it works perfectly fine if content of `customconfig` is just in `config` file and I just do `./build-docker.sh`.

samtygier, 2019-02-16:
Sorry, I didn't test the docker version of this hard enough in 7fbfdda31e

`build-docker.sh` uses 2 methods to pass in the config, first it passes it as in the docker run command with `--env-file`, then secondly by passing the `config` file into the container. My patch added a second `--env-file`, but it turns the values can be overwritten by the `config` file.

A better method is to remove the `--env-file` method completely, and just pass any arguments given to `build-docker.sh` through to `build.sh`

Patch to follow.

PR #259 (2019-02-18): Ensure extra config file is passed to docker
Fixes #253

Rather than use --env-file, which gets overwritten by the regular config file, pass the any arguments to build-docker.sh through to build.sh.
commit c147df99 (2019-02-18): Ensure extra config file is passed to docker

Fixes #253

Rather than use --env-file, which gets overwritten by the regular config file, pass the any arguments to build-docker.sh through to build.sh.
closed 2019-02-18 by XECDesign