RPi-Distro/pi-gen #257: Docker: losetup failed to set up loop device: No such file or directory

samtygier, 2019-02-16:
When building with docker I am getting the error:

`losetup: /pi-gen/work/2019-02-13-Raspbian/export-image/2019-02-13-Raspbian-lite.img: failed to set up loop device: No such file or directory`

even though that file does exist.

To reproduce, make docker build stop at final rsync in export-image/prerun.sh . Loop devices are at kernel level outside the docker container, so they persist after the failure. They can be seen with the command `losetup --list`

The unmount_image function in scripts/common uses:
LOOP_DEVICES=$(losetup -j "${1}" | cut -f1 -d':')
but does not see the old loop devices, because the path is missing the "/pi-gen/work", so it is unable to remove them. I imagine the path difference is to do with the how the container views the filesystem. This can be solved by listing all the loop devices, and matching by filename (patch to follow).

There are a few older closed bugs mentioning the issue, #104 #193

closed 2019-02-18 by XECDesign

jnaulty, 2019-03-20:
I ran into this issue when building with docker.
What was the solution?

I'm currently building from master at commit: https://github.com/RPi-Distro/pi-gen/commit/30a1528ae13f993291496ac8e73b5ac0a6f82585

_tail of build-docker.sh_
``` 
Device                                                                              Boot Start     End Sectors  Size Id Type
/pi-gen/work/2019-03-20-rpi-security/export-image/2019-03-20-rpi-security-lite.img1       8192   97459   89268 43.6M  c W95 FAT32 (LBA)
/pi-gen/work/2019-03-20-rpi-security/export-image/2019-03-20-rpi-security-lite.img2      98304 3522559 3424256  1.6G 83 Linux

Command (m for help): The partition table has been altered.
Syncing disks.

losetup: /pi-gen/work/2019-03-20-rpi-security/export-image/2019-03-20-rpi-security-lite.img: failed to set up loop device: No such file or directory
```

samtygier, 2019-03-20:
Does it happen every time? I have seen it occasionally since the above patches but not enough to debug.

Maybe it is a bug in `losetup`. The error message is certainly wrong as the file mentioned does exist.

I wonder if the solution is to avoid using losetup. We could just `mount` the image file directly but that does not work with partitions. So we would have to make an image file for each partition and cat them together at the end. Another option might be mounting with `guestmount`.

jnaulty, 2019-03-20:
yes, the issue is consistent on my system (Using Docker Build, but running on an Ubuntu 18.04 host)

I'm going to spend a little to debugging later

jnaulty, 2019-03-20:
I add a `set -x` to the `prerun.sh` script in `export-image/`
```
+ ROOT_LENGTH=1753219072
++ losetup --show -f -o 4194304 --sizelimit 45705216 /pi-gen/work/2019-03-20-rpi-security/export-image/2019-03-20-rpi-security-lite-qemu.img
+ BOOT_DEV=/dev/loop25
++ losetup --show -f -o 50331648 --sizelimit 1753219072 /pi-gen/work/2019-03-20-rpi-security/export-image/2019-03-20-rpi-security-lite-qemu.img
losetup: /pi-gen/work/2019-03-20-rpi-security/export-image/2019-03-20-rpi-security-lite-qemu.img: failed to set up loop device: No such file or directory
+ ROOT_DEV=

real    20m7.451s
user    0m0.285s
sys     0m0.280s

```
I logged spun up a container for debugging:
`sudo docker run -it --privileged --volumes-from=pigen_work pi-gen /bin/bash`

Running the `losetup` command manually  for the root_dev worked:
```
root@2cc431bf79bc:/# losetup --show -f -o 50331648 --sizelimit 1753219072 /pi-gen/work/2019-03-20-rpi-security/export-image/2019-03-20-rpi-security-lite-qemu.img
/dev/loop26
root@2cc431bf79bc:/# exit
```

jnaulty, 2019-03-20:
Also, I am noticing that these loopback devices are not detached upon exit. We  might want to add some logic to do that if the script is interrupted or exits abruptly.

jnaulty, 2019-03-20:
Ok, it's not so consistent. It made it past this failure the third time, but was failed at a later stage 
```
mkfs.fat: warning - lowercase labels might not work properly with DOS or Windows
mkdosfs: Too few blocks for viable filesystem
```
:thinking:

jnaulty, 2019-03-21:
my issue was resolved by this comment: https://github.com/RPi-Distro/pi-gen/issues/105#issuecomment-326754980

remove the `EXPORT*` files in stage 5

samtygier, 2020-01-08:
Looks like not having the `loop` module loaded can cause this.

KeithPetro, 2020-01-16:
> Looks like not having the `loop` module loaded can cause this.

I am experiencing the same issue and `loop` is loaded as a builtin module for me.

PR #258 (2019-02-18): Make sure unmount_image finds stale loop devices
Fixes #257 #104 #193

Instead of searching by full path, which is prone to fail, read full list and grep on filename.
commit 7068086c (2019-02-18): Make sure unmount_image finds stale loop devices

Fixes #257 #104 #193

Instead of searching by full path, which is prone to fail, read full list and grep on filename.