RPi-Distro/pi-gen #264: Module binfmt_misc not loaded in host

erm3nda, 2019-03-06:
OS: Manajaro, kernel 4.19.

I tried to modprobe binfmt_misc and seems loaded.

Tried also with Manjaro package `manjaro-arm-tools` instead of the `qemu-user-static`.
Tried both `./build.sh` and `./build-docker.sh`. Both commands output the same problem.
Tried AUR package `qemu-user-static-bin` and got the same result.

Listed the modules, it is present, it's loaded with no warning ... no way.

XECDesign, 2019-03-06:
Could you please provide more information, including exact output and how you've determine binfmt_misc is in fact loaded.  What's the output of lsmod?

erm3nda, 2019-03-06:
Hi, thank you.

Exact error output:

```
sudo ./build.sh
[sudo] password for erm3nda: 
Module binfmt_misc not loaded in host
Please run:
  sudo modprobe binfmt_misc
```

Determined by the "Please run" suggestion: 
`sudo modprobe binfmt_misc`.

As far as i do understand modprobe returns nothing when is ok.

XECDesign, 2019-03-06:
> What's the output of lsmod?

erm3nda, 2019-03-06:
There's no `binfmt_misc`. Is modprobe liying?

```
Module                  Size  Used by
veth                   28672  0
nls_utf8               16384  0
isofs                  45056  0
xfs                  1634304  0
jfs                   208896  0
btrfs                1372160  0
xor                    24576  1 btrfs
raid6_pq              122880  1 btrfs
raid1                  45056  1
md_mod                167936  2 raid1
uas                    28672  1
usb_storage            73728  1 uas
usblp                  24576  0
ipt_MASQUERADE         16384  1
nf_conntrack_netlink    49152  0
nfnetlink              16384  2 nf_conntrack_netlink
xfrm_user              45056  1
xfrm_algo              16384  1 xfrm_user
iptable_nat            16384  1
nf_nat_ipv4            16384  2 ipt_MASQUERADE,iptable_nat
xt_addrtype            16384  2
iptable_filter         16384  1
xt_conntrack           16384  1
nf_nat                 36864  1 nf_nat_ipv4
nf_conntrack          163840  5 xt_conntrack,nf_nat,ipt_MASQUERADE,nf_nat_ipv4,nf_conntrack_netlink
nf_defrag_ipv6         20480  1 nf_conntrack
nf_defrag_ipv4         16384  1 nf_conntrack
libcrc32c              16384  4 nf_conntrack,nf_nat,btrfs,xfs
br_netfilter           24576  0
bridge                192512  1 br_netfilter
stp                    16384  1 bridge
llc                    16384  2 bridge,stp
overlay               126976  0
aufs                  307200  0
ccm                    20480  9
cmac                   16384  1
rfcomm                 86016  4
fuse                  118784  7
snd_hda_codec_hdmi     57344  1
snd_hda_codec_realtek   118784  1
snd_hda_codec_generic    86016  1 snd_hda_codec_realtek
bnep                   24576  2
arc4                   16384  2
i915                 2097152  32
btusb                  53248  0
bbswitch               16384  0
btrtl                  16384  1 btusb
btbcm                  16384  1 btusb
snd_soc_skl           118784  0
btintel                24576  1 btusb
snd_soc_skl_ipc        73728  1 snd_soc_skl
snd_soc_sst_ipc        16384  1 snd_soc_skl_ipc
intel_rapl             24576  0
bluetooth             655360  31 btrtl,btintel,btbcm,bnep,btusb,rfcomm
snd_soc_sst_dsp        36864  1 snd_soc_skl_ipc
x86_pkg_temp_thermal    16384  0
snd_hda_ext_core       28672  1 snd_soc_skl
intel_powerclamp       16384  0
snd_soc_acpi_intel_match    24576  1 snd_soc_skl
coretemp               16384  0
snd_soc_acpi           16384  2 snd_soc_acpi_intel_match,snd_soc_skl
iwlmvm                442368  0
snd_soc_core          270336  1 snd_soc_skl
kvmgt                  28672  0
vfio_mdev              16384  0
kvm_intel             245760  0
[... 100 lines cut ...]
x_tables               45056  5 xt_conntrack,iptable_filter,ipt_MASQUERADE,xt_addrtype,ip_tables
hid_generic            16384  0
usbhid                 57344  0
hid                   139264  4 i2c_hid,usbhid,hid_multitouch,hid_generic
ext4                  737280  1
crc32c_generic         16384  0
crc16                  16384  2 bluetooth,ext4
mbcache                16384  1 ext4
jbd2                  126976  1 ext4
fscrypto               32768  1 ext4
sd_mod                 61440  4
ahci                   40960  1
serio_raw              16384  0
libahci                40960  1 ahci
atkbd                  32768  0
libps2                 16384  2 atkbd,psmouse
libata                278528  2 libahci,ahci
xhci_pci               16384  0
crc32c_intel           24576  3
scsi_mod              253952  5 sd_mod,usb_storage,uas,libata,sg
xhci_hcd              266240  1 xhci_pci
i8042                  32768  0
serio                  28672  6 serio_raw,atkbd,psmouse,i8042

```

XECDesign, 2019-03-06:
When you run modprobe do any errors show up in dmesg? Does Manjaro maybe build binfmt_misc into the kernel rather than as a module?

erm3nda, 2019-03-06:
There's no dmesg output related to modprobe.

> Does Manjaro maybe build binfmt_misc into the kernel rather than as a module?

Makes a bit of sense, but i don't understand why should it make a difference, cuz the support is enabled.
I sent a query to #manjaro to see if anyone could say me that.

Thanks.

erm3nda, 2019-03-06:
> Does Manjaro maybe build binfmt_misc into the kernel rather than as a module?

https://gitlab.manjaro.org/search?utf8=%E2%9C%93&search=binfmt&group_id=&project_id=585&search_code=true&repository_ref=master

Yes, binfmt_misc support builtin in Kernel.

XECDesign, 2019-03-06:
What's the output of `cat /proc/mounts`? Is there a line that looks like this?
```
binfmt_misc /proc/sys/fs/binfmt_misc binfmt_misc rw,relatime 0 0
```

The check could probably be updated to handle this case.

samtygier, 2019-03-06:
That's probably a better way to check. From the Manjaro live cd I get

```
mount | grep binfmt
systemd-1 on /proc/sys/fs/binfmt_misc type autofs (rw,relatime,fd=33,pgrp=1,timeout=0,minproto=5,maxproto=5,direct,pipe_ino=9696)
```

So in scripts/dependencies_check the check should maybe be:
```
if ! mount | grep "/proc/sys/fs/binfmt_misc" >/dev/null
```

XECDesign, 2019-03-06:
I think `grep -q "/proc/sys/fs/binfmt_misc" /proc/mounts` is a bit cleaner.

Alright, I'll add that.

closed 2019-03-06 by XECDesign

erm3nda, 2019-03-06:
Yes.

```
mount | grep binfmt
systemd-1 on /proc/sys/fs/binfmt_misc type autofs (rw,relatime,fd=39,pgrp=1,timeout=0,minproto=5,maxproto=5,direct,pipe_ino=487)
binfmt_misc on /proc/sys/fs/binfmt_misc type binfmt_misc (rw,relatime)
```

So, fixed :-)

Many thanks.

smg-bg, 2019-07-10:
For some reason the change does not work for me. lsmod returns the module as loaded, but mount is not returning anything about binfmt_misc.

Any ideas?

XECDesign, 2019-07-12:
what distro?

smg-bg, 2019-07-12:
It's running inside a docker image based on Ubuntu:18.04, which run on Ubuntu. If I remove this check it seems to build perfectly fine. 

Maybe both checks _lsmod_ and _mount_ need to be combined with _or_?

XECDesign, 2019-07-23:
Is the issue that the mount exists on the host system but not from inside the docker container? If that's the case, I'm not sure what the right solution is.

Athwale, 2021-11-21:
Hi I am having a similar problem. I am attempting to build a rpi image with the build-docker.sh on my 64bit Fedora 34.
Running it gives me:
```
$ ./build-docker.sh 
Sending build context to Docker daemon  889.3kB
Step 1/6 : ARG BASE_IMAGE=debian:bullseye
Step 2/6 : FROM ${BASE_IMAGE}
 ---> 59b34e04dc27
Step 3/6 : ENV DEBIAN_FRONTEND noninteractive
 ---> Using cache
 ---> 8b9ea18dc7b5
Step 4/6 : RUN apt-get -y update &&     apt-get -y install --no-install-recommends         git vim parted         quilt coreutils qemu-user-static debootstrap zerofree zip dosfstools         libarchive-tools libcap2-bin rsync grep udev xz-utils curl xxd file kmod bc        binfmt-support ca-certificates qemu-utils kpartx     && rm -rf /var/lib/apt/lists/*
 ---> Using cache
 ---> 8228902b86d4
Step 5/6 : COPY . /pi-gen/
 ---> Using cache
 ---> bc784b57727c
Step 6/6 : VOLUME [ "/pi-gen/work", "/pi-gen/deploy"]
 ---> Using cache
 ---> 9f92c61a4b7a
Successfully built 9f92c61a4b7a
Successfully tagged pi-gen:latest
WARNING: The requested image's platform (linux/386) does not match the detected host platform (linux/amd64) and no specific platform was requested
Module binfmt_misc not loaded in host
Please run:
  sudo modprobe binfmt_misc

real	0m0.509s
user	0m0.022s
sys	0m0.028s
```

My Fedora laptop does have binfmt_misc in the output of:
```
$ lsmod | grep binfmt_misc
binfmt_misc            24576  1
```
But this seems to be a problem with the docker environment. Because if I comment out the `exit 1` on line 50 in the scripts/dependencies_check the image gets built seemingly ok.  
Unfortunately I currently do not have a RPI to test that the image runs, but at least it gets saved in the deploy directory.

commit 67241b96 (2019-03-06): Handle cases where binfmt_misc is built into the kernel

Fixes #264