RPi-Distro/pi-gen #265: pxz is no longer available for debian buster

plus2equalsme, 2019-03-07:
With the release of debian buster approaching, I decided to attempt running pi-gen on my buster system. At this point I discovered pxz has been obsoleted by a recent release of xz, and as such has already been removed from the buster repos and will be removed from sid (and all following versions of debian soon).  According to [https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=919809](https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=919809) running `xz -T` accomplishes the same as running `pxz`.

XECDesign, 2019-03-07:
Thanks for the heads up. I'll give `xz -T` a go.

zoff99, 2019-03-08:
https://linux.die.net/man/1/xz
says:

```
Multithreaded compression and decompression are not implemented yet,
    so this option has no effect for now.
```

is that no longer correct?

ThiagoVinicius, 2019-03-08:
@zoff99

Please refer to https://git.tukaani.org/?p=xz.git;a=blob;f=NEWS;hb=HEAD

Search for `multi-thread`, there.

You may also confirm that the multi-thread capable version is on Debian Buster, by checking the package version here: https://packages.debian.org/buster/xz-utils

andig, 2019-03-10:
@plus2equalsme A little OT, please excuse the side question.
How would I modify the (docker) build process to use buster? Are you doing anything else than using buster as runtime environment for the build? I'd prefer to do this for sake of php 7.3. Is it enough to change the docker base image or are there changes in multiple places?

plus2equalsme, 2019-03-10:
> @plus2equalsme A little OT, please excuse the side question.
> How would I modify the (docker) build process to use buster? Are you doing anything else than using buster as runtime environment for the [...]

@andig I do not use docker at all, however, I will answer as best as I can. All I did was clone the repo and try to install all the dependencies onto my laptop which is running Debian Buster as my OS. Because I could not install pxz, I could not run the script, but if I had, it would have still built raspbian stretch. If you want to run it on a buster based system, and still build stretch, I *think* all you would have to do is edit the depends file to remove pxz and not let the script run the NOOBS build.

If you want to try to build raspbian buster, on the other hand, there are additional changes you would need to make. It is my understanding that people have been able to upgrade to buster from a running stretch install, but I'm not sure the repo is in a state that it is worth trying to build an image yet. Talking through either of those is way offtopic for this thread though.

XECDesign, 2019-03-11:
Should be fixed by https://github.com/RPi-Distro/pi-gen/commit/fb23b8ba92ffc893fe257375ce79e56d3dbe7504

plus2equalsme, 2019-03-11:
I'm running a standard lite build right now. However, as it runs I'm seeing a lot of Errors in the realm of can't open shared object file on the kernel modules/drivers (all .so files). I can get you the output if you want when it's done, do you want it just pasted here, or some other method?

XECDesign, 2019-03-11:
Here is fine

closed 2019-03-12 by XECDesign