RPi-Distro/pi-gen #266: Question: initialise database user/schema during image build?

andig, 2019-03-11:
I hope this question is ok to ask in the issue tracker. 

I need to prepare database user/schema for the application to work. I could leave this to a run-once script but it seams better to do this during image build for sake of error handling/ testing.

Is it possible to do that in the chroot environment (of course given e.g. default-mysql-server has been installed)?

XECDesign, 2019-03-11:
I'm not 100% sure. Since it's running in chroot, systemd won't automatically start services, so the SQL server wouldn't be running. I'm sure there's a way to handle this, but it's not something I've tried. You may need to start the service, prepare your DB, then stop the service. Or is there an 'offline' way to do this?

andig, 2019-03-11:
Doesn't seem to be that easy, sigh. Adding:

    on_chroot << EOF
    /usr/sbin/mysqld
    EOF

gives:

    2019-03-11 13:46:04 4104719152 [Note] /usr/sbin/mysqld (mysqld 10.1.37-MariaDB-0+deb9u1) starting as process 246 ...

...and then the build stops hard.

XECDesign, 2019-03-11:
Is it meant to automatically background itself?

If it's running in the foreground, then it makes sense that it would block everything. I was thinking that you should start it via the systemd service, not directly.

andig, 2019-03-11:
Mhhm, from the build `/var/log/mysql/error.log`:

    2019-03-11 13:52:45 4104719152 [ERROR] Can't start server : Bind on unix socket: No such file or directory
    2019-03-11 13:52:45 4104719152 [ERROR] Do you already have another mysqld server running on socket: /var/run/mysqld/mysqld.sock ?
    2019-03-11 13:52:45 4104719152 [ERROR] Aborting

So maybe it *is* running. I'll do some more tests...

XECDesign, 2019-03-11:
Make sure it's not running on your host os.

andig, 2019-03-11:
I'm not enough unix pro to get this done easily. Could you give me a small hint how to run commands inside the chroot environment from an interactive bash inside the container?

    docker run -it --privileged --volumes-from=pigen_work pi-gen /bin/bash

This doesn't work as expected:

    root@f741ef6a6699:/# export ROOTFS_DIR=/pi-gen/work/2019-03-11-volkszaehler/volkszaehler/rootfs/
    root@f741ef6a6699:/# source /pi-gen/scripts/common
    root@f741ef6a6699:/# on_chroot < /bin/bash -i

The stage `volkszaehler` exists and has a rootfs initialized from previous step. Any hint welcome- hope I'm not overdoing it for a gh issue?

XECDesign, 2019-03-11:
`systemd-nspawn -D /pi-gen/work/2019-03-11-volkszaehler/volkszaehler/rootfs/`

andig, 2019-03-11:
Not really (any stage):

    root@0947fe2b1d27:/# systemd-nspawn -D /pi-gen/work/2019-03-11-volkszaehler/stage2/rootfs/
    Spawning container rootfs on /pi-gen/work/2019-03-11-volkszaehler/stage2/rootfs.
    Press ^] three times within 1s to kill container.
    Failed to open system bus: No such file or directory
    Attempted to remove disk file system, and we can't allow that.

Guess I'll have to give up as this is beyond my skills :/

XECDesign, 2019-03-11:
Haven't seen before, but I guess that's because I don't use docker.

You could try just `chroot /pi-gen/work/2019-03-11-volkszaehler/volkszaehler/rootfs/`

andig, 2019-03-12:
No chance. Mysqld can't bind to socket (the socket folder is empty):

    export ROOTFS_DIR=/pi-gen/work/2019-03-11-volkszaehler/volkszaehler/rootfs/
    mount -t proc proc "${ROOTFS_DIR}/proc"
    mount --bind /dev "${ROOTFS_DIR}/dev"
    mount --bind /dev/pts "${ROOTFS_DIR}/dev/pts"
    mount --bind /sys "${ROOTFS_DIR}/sys"
    chroot ${ROOTFS_DIR}

This is the output of `/usr/sbin/mysqld`:

    2019-03-12  9:27:47 4104719152 [Note] InnoDB: Using mutexes to ref count buffer pool pages
    2019-03-12  9:27:47 4104719152 [Note] InnoDB: The InnoDB memory heap is disabled
    2019-03-12  9:27:47 4104719152 [Note] InnoDB: Mutexes and rw_locks use GCC atomic builtins
    2019-03-12  9:27:47 4104719152 [Note] InnoDB: GCC builtin __atomic_thread_fence() is used for memory barrier
    2019-03-12  9:27:47 4104719152 [Note] InnoDB: Compressed tables use zlib 1.2.8
    2019-03-12  9:27:47 4104719152 [Note] InnoDB: Using Linux native AIO
    2019-03-12  9:27:47 4104719152 [Note] InnoDB: Using generic crc32 instructions
    qemu: Unsupported syscall: 243
    2019-03-12 09:27:47 f4a90b30 InnoDB: Error: Linux Native AIO interface is not supported on this platform. Please check your OS documentation and install appropriate binary of InnoDB.
    InnoDB: You can disable Linux Native AIO by setting innodb_use_native_aio = 0 in my.cnf
    2019-03-12  9:27:47 4104719152 [Warning] InnoDB: Linux Native AIO disabled.
    2019-03-12  9:27:47 4104719152 [Note] InnoDB: Initializing buffer pool, size = 128.0M
    2019-03-12  9:27:47 4104719152 [Note] InnoDB: Completed initialization of buffer pool
    2019-03-12  9:27:47 4104719152 [Note] InnoDB: Highest supported file format is Barracuda.
    2019-03-12  9:27:47 4104719152 [Note] InnoDB: 128 rollback segment(s) are active.
    2019-03-12  9:27:47 4104719152 [Note] InnoDB: Waiting for purge to start
    2019-03-12  9:27:47 4104719152 [Note] InnoDB:  Percona XtraDB (http://www.percona.com) 5.6.41-84.1 started; log sequence number 1616827
    2019-03-12  9:27:47 3477074752 [Note] InnoDB: Dumping buffer pool(s) not yet started
    2019-03-12  9:27:47 4104719152 [Note] Plugin 'FEEDBACK' is disabled.
    2019-03-12  9:27:47 4104719152 [Note] Server socket created on IP: '127.0.0.1'.
    2019-03-12  9:27:47 4104719152 [ERROR] Can't start server : Bind on unix socket: Permission denied
    2019-03-12  9:27:47 4104719152 [ERROR] Do you already have another mysqld server running on socket: /var/run/mysqld/mysqld.sock ?
    2019-03-12  9:27:47 4104719152 [ERROR] Aborting

If this is a permission problem then `sudo` doesn't seem to help either:

    # sudo
    Unknown host QEMU_IFLA type: 40
    Unknown host QEMU_IFLA type: 41
    Unknown host QEMU_IFLA type: 40
    Unknown host QEMU_IFLA type: 41
    Unknown QEMU_IFLA_INFO_KIND ipip
    Unknown host QEMU_IFLA type: 40
    Unknown host QEMU_IFLA type: 41
    Unknown QEMU_IFLA_INFO_KIND ip6tnl
    Unknown host QEMU_IFLA type: 40
    Unknown host QEMU_IFLA type: 41
    sudo: unable to resolve host 4def99a380c9: No such file or directory

Seems like the combination of docker/qemu etc doesn't really work in this case.

XECDesign, 2019-03-12:
Hmm. Not familiar with the ins and outs of mariadb or docker, so I can't be of much help here.

XECDesign, 2019-04-15:
Closing since there hasn't been much activity lately and it's more of a generic mariadb question.

closed 2019-04-15 by XECDesign

zoic21, 2020-06-03:
For information you can fix this issue by doing that in chroot : 

```
mkdir /run/mysqld
chmod 777 -R /run/mysqld
```

And start mysql with service (not by systemctl)