RPi-Distro/pi-gen #267: Unable to locate package raspi-copies-and-fills

smg-bg, 2019-03-11:
I was able to build a distro as of March 1st, but today without changing the source version I'm getting this error. I'm not sure how I can check if the package is available on http://raspbian.raspberrypi.org/raspbian, but I assume it just disappeared?! Or I'm totally wrong :)

> [18:28:45] Begin /vsts/agent/_work/1/s/pi-gen/stage2
> [18:28:45] Begin /vsts/agent/_work/1/s/pi-gen/stage2/prerun.sh
> [18:28:50] End /vsts/agent/_work/1/s/pi-gen/stage2/prerun.sh
> [18:28:50] Begin /vsts/agent/_work/1/s/pi-gen/stage2/00-copies-and-fills
> [18:28:50] Begin /vsts/agent/_work/1/s/pi-gen/stage2/00-copies-and-fills/01-packages
> Reading package lists...
> Building dependency tree...
> Reading state information...
> E: Unable to locate package raspi-copies-and-fills
> ##[error]Bash exited with code '100'.
> ##[section]Finishing: Build Image

Maybe someone will have any idea?

XECDesign, 2019-03-11:
The package was broken and had to be pulled asap, since it broke raspbian when people upgraded. The commit responsible for the breakage has been removed and the package re-uploaded.

It will be available again within an hour, when the load balancers sync up.

smg-bg, 2019-03-11:
Thanks for the quick reply. Much appreciated!

XECDesign, 2019-03-11:
It should be back up now.

admindotnu, 2019-03-11:
Breakes at the moment pi-gen

ERROR: ld.so: object '/usr/lib/arm-linux-gnueabihf/libarmmem-${PLATFORM}.so' from /etc/ld.so.preload cannot be preloaded (cannot open shared object file): ignored.

XECDesign, 2019-03-11:
Safe to ignore. Qemu has patches that will eventually get rid of the problem, but for now I'll find a way to silence them while pi-gen is running.

admindotnu, 2019-03-12:
make ${PLATFORM} var available in env ?

admindotnu, 2019-03-12:
Ah ok, clear. Thank you, learned something after long time new ;-)

closed 2019-03-13 by XECDesign