RPi-Distro/pi-gen #268: /etc/ld.so.preload cannot be preloaded

caromeo, 2019-03-14:
I came across this error:
`ERROR: ld.so: object '/usr/lib/arm-linux-gnueabihf/libarmmem-${PLATFORM}.so' from /etc/ld.so.preload cannot be preloaded (cannot open shared object file): ignored.`

 while building the image in the last few weeks. I built it many other times but only recently I've had this problem. No working solution found till now...

XECDesign, 2019-03-14:
https://github.com/RPi-Distro/pi-gen/issues/267#issuecomment-471735794

People are reporting other issues with that package, so the message should go away tomorrow.

FredericGuilbault, 2019-03-14:
Is it safe to ignore or it have impact ? 

The binary build of nodejs packages have broke at the same time for me.
 It can be a coincidence ...or not. ***paranoïa intensifies***

XECDesign, 2019-03-14:
No impact. What's happening is that qemu isn't setting that PLATFORM auxval, so when ld.so tries to find the file, it fails and ignores it.

https://bugs.launchpad.net/qemu/+bug/1813034

FredericGuilbault, 2019-03-16:
Im sorry to comeback with this agan but I was compiling `xygrib` without problem before this event and now it broke (this and others binaries build ) 

This is the revelant part of my pi-gen build log : 
```
[20:54:29] Begin /var/www/lysmarine-gen/91-xygrib/03-xygrib/02-run.sh

ERROR: ld.so: object '/usr/lib/arm-linux-gnueabihf/libarmmem-.so' from /etc/ld.so.preload cannot be preloaded (cannot open shared object file): ignored.

System is unknown to cmake, create: Linux to use this system, please send your config file to cmake@www.cmake.org so it can be added to cmake

Your CMakeCache.txt file was copied to CopyOfCMakeCache.txt. Please send that file to cmake@www.cmake.org.

CMake Error at CMakeLists.txt:45 (find_package):

  Could not find a package configuration file provided by "Qt5Widgets" with
  any of the following names:

    Qt5WidgetsConfig.cmake
    qt5widgets-config.cmake

  Add the installation prefix of "Qt5Widgets" to CMAKE_PREFIX_PATH or set
  "Qt5Widgets_DIR" to a directory containing one of the above files.  If
  "Qt5Widgets" provides a separate development package or SDK, be sure it has
  been installed.

-- Configuring incomplete, errors occurred!
```
And when I do a `find ./ -name Qt5WidgetsConfig.cmake`  it says :
```
./V0.8.0-lysmarine/91-xygrib/rootfs/usr/lib/arm-linux-gnueabihf/cmake/Qt5Widgets/Qt5WidgetsConfig.cmake
```
So the file is here that's not the problem. 

Maybe cmake need that `${PLATFORM}` variable to know where to find  the .cmake files ?  
Im making a guess here, but really this is out of my knowlege scope :)

FredericGuilbault, 2019-03-16:
The error is gone this morning and magicaly it's compiling agan. 

Is it gone for you too @caromeo ?

caromeo, 2019-03-16:
@FredericGuilbault .... yes. We can close the issue

closed 2019-03-17 by caromeo