RPi-Distro/pi-gen #269: set apt for DEBIAN_FRONTEND=noninteractive

FredericGuilbault, 2019-03-15:
When there is multiple users created, Wicd package prompt a ncurse dialog.

[image]

 
So I have to move the install process to a run.sh file and go like this: 
```
on_chroot << EOF
export DEBIAN_FRONTEND=noninteractive
apt install -y wicd wicd-curses wicd-gtk
EOF
```

I can't figure a use case wher pi-gen would need user-interaction during the compiling. Maybe it would be a good improvement to set the `export DEBIAN_FRONTEND=noninteractive` variable by default.

XECDesign, 2019-03-15:
The idea was that the build should interrupt so that I can go back and preseed whatever new debconf options come up. Otherwise packages could make assumptions and I wouldn't know what they are.

FredericGuilbault, 2019-03-15:
good point.

closed 2019-03-15 by FredericGuilbault