RPi-Distro/pi-gen #274: Feature request: Micro:bit udev rule

gbaman, 2019-04-04:
With the new version of the Chromium package (72) now shipping, it brings with it proper WebUSB support. Although this is a protocol used by very very few devices, it is used by the BBC Micro:bit and makes quite a difference with the speed kids can develop using it.

Both Microsoft Makecode the web based Python editor support WebUSB pairing, speeding everything up vs downloading .hex files and copying them to the Micro:bit flash drive.

There is though a single blocker for the average user, a udev rule needs added in. 

**I would propose this udev rule be included in stock Raspbian to remove that blocker.**
The needed rule (that would be specific to only Micro:bits) would be 
```SUBSYSTEM=="usb", ATTR{idVendor}=="0d28", MODE="0664", GROUP="plugdev"```
(taken straight from the [Micro:bit website](https://support.microbit.org/support/solutions/articles/19000084059-beta-testing-web-usb))

I can confirm with that rule added, it works great!
[image]

Thoughts?

XECDesign, 2019-04-04:
First time hearing of WebUSB. Yeah, why not. I can put it into raspberrypi-sys-mods.

gbaman, 2019-04-04:
Excellent, thanks @XECDesign. Will hopefully simply make it even more seamless for kids with micro:bits and Pis.

gbaman, 2019-07-27:
@XECDesign did this one get anywhere for integration into Raspbian?

XECDesign, 2019-07-29:
It dropped too far down the todo list. I've just added it for the next raspberrypi-sys-mods release: https://github.com/RPi-Distro/raspberrypi-sys-mods/commit/761fca6da00754786000dbf54c7113fe6f4215a2

gbaman, 2019-07-29:
Fantastic, having it in place for late August would be perfect for schools getting cards ready for the coming school.

XECDesign, 2019-07-30:
`lintian` just threw this at me: https://lintian.debian.org/tags/udev-rule-missing-uaccess.html

Does `SUBSYSTEM=="usb", ATTR{idVendor}=="0d28", TAG+="uaccess"` work?

XECDesign, 2019-08-01:
I managed to find a micro:bit to test it on. With the uaccess rule I am able to pair with it, but downloading code to it seems to time out. I tried the plugdev + 644 rule and I see the same behaviour, so I'll just add the uaccess rule for now.

gbaman, 2019-08-01:
Odd, I will give it another look this evening, maybe Buster broke it...

XECDesign, 2019-08-02:
The `uaccess` version of the rule is in the latest sys-mods package. In my brief testing it worked no worse than the `plugdev` rule. Re-open if you get it to work better with another rule .

closed 2019-08-02 by XECDesign