RPi-Distro/pi-gen #275: Can we reduce diskspace needs using hardlinks/cow-shell?

iKarith, 2019-04-09:
I've been working on a Debian + Raspbian build tool based loosely upon Ubuntu's {p,cow}builder-dist tool because frustratingly it doesn't already exist, and it's got me wondering about cowdancer and pi-gen.

A lot of time and disk space gets used by pi-gen duplicating each stage from its parent. Each stage after 0 begins by calling this in its prerun.sh script:

```bash
copy_previous(){
	if [ ! -d "${PREV_ROOTFS_DIR}" ]; then
		echo "Previous stage rootfs not found"
		false
	fi
	mkdir -p "${ROOTFS_DIR}"
	rsync -aHAXx --exclude var/cache/apt/archives "${PREV_ROOTFS_DIR}/" "${ROOTFS_DIR}/"
}
export -f copy_previous
```

That'd go a lot faster if it were copying hardlinks (which rsync can do, though the manpage is a bit misleading about what `DEST` in in the context of `--link-dest=DEST`. Of course the reason you don't already do that is later stages can corrupt earlier ones if earlier stages are all hardlinked.

This is what cow-shell does. :) If you have a hardlink tree, cow-shell guarantees copy-on-write, meaning if stage 2 writes to a file in a rootfs hardlinked to a stage 1 file, the data is copied and the link severed before a write happens.

Some complexity would be added to pi-gen to support this—namely commands that may write to the system should be invoked through cow-shell. But then, it'd save 2.4GB in duplicates from of stage 0 alone.

(The other speedup I could suggest would be saving and restoring the apt cache. I'll submit a patch for that if desired since it requires less discussion.)

A clean single build of Raspbian creates a work directory of 32 GB at the time of writing.

XECDesign, 2019-04-09:
I was going to use branching qcow images instead, the way spindle did it, but this approach sounds better.

If it produces identical files in the end, I have no problem with saving time and disk space. A PR would be much appreciated.

iKarith, 2019-04-10:
qcow also works, but I don't have as much knowledge of how to set that up. You'd replaced spindle with pi-gen by the time I was becoming interested in that sort of thing, so I didn't really look into it too much.

The thing I _really_ wish I could take the time to wrap my head around is how Debian's setup works and what it'd take to adapt that for Raspbian. AFAICT the stuff you use to generate Debian+PIXEL images isn't actually released anywhere at the moment. Not 100% sure it'd help if it was—I did notice that your release installs i386 packages (only) with an amd64 kernel, I guess for better hardware support on more modern systems? That breaks BTW if you start doing things like dkms, as someone who tried to use it because the only Debian stuff they were familiar with was Raspbian discovered the hard way. You do have amd64 versions of the PIXEL packages, so I'd have rolled him an amd64 image real fast if I had your build script customization and about as much documentation as we have for pi-gen to work with. :)

Anyway, I'll take another look at spindle and see if I can work out what it's doing and see if we can bring over its qcow images more easily than the cowdancer stuff. It's likely to be a smaller patch.

iKarith, 2019-04-10:
Actually upon closer look, the changes could be fairly minimal to use cowdancer. The steps (which I shall not attempt now as I'm currently sleeping on nearly UK time at the moment) are these:

1. Migrate `run_sub_stage` to its own script and ensure that it runs that way.
2. Change everything so run_sub_stage's CWD is in the work directory because cow-shell is somewhat braindead about its .ilist file location and ignores the environment variable *sigh*
3. Run `run_sub_stage` through cow-shell.
4. Modify copy_previous to use hardlinks.
5. Verify the result.

It's possible to invoke individual commands using cow-shell in which case not deleting the .ilist file between actually makes sense, but they need to be invoked inside the directory tree that's going to be changed. We could probably patch that for buster (if 0.86 didn't already fix it), but that's unlikely to be backported to stretch.

Haven't spent more than a minute looking at spindle yet.