RPi-Distro/pi-gen #276: Unable to build image

dernasherbrezon, 2019-04-14:
Hi,

Trying to build an image directly from the latest tag and getting this:

> [10:05:19] Begin /home/pi/pi-gen/stage0/prerun.sh
> E: No such script: /debootstrap/scripts/stretch
> [10:05:20] End /home/pi/pi-gen/stage0/prerun.sh
> [10:05:20] Begin /home/pi/pi-gen/stage0/00-configure-apt
> [10:05:20] Begin /home/pi/pi-gen/stage0/00-configure-apt/00-run.sh
> install: target '/home/pi/pi-gen/work/2019-04-14-Raspbian/stage0/rootfs/etc/apt/' is not a directory: No such file or directory

Tried to look through the scripts and haven't found any obvious debootstrap issues.

> pi@raspberrypi:~/pi-gen $ apt list -a debootstrap
> Listing... Done
> debootstrap/stable,now 1.0.89 all [installed]

Looks like path is broken?

> pi@raspberrypi:~/pi-gen $ ls -lh /usr/share/debootstrap/scripts/stretch 
> lrwxrwxrwx 1 root root 3 Mar 10  2017 /usr/share/debootstrap/scripts/stretch -> sid

Running on stretch, trying to build stretch. Looks valid:

> pi@raspberrypi:~/pi-gen $ lsb_release -a
> No LSB modules are available.
> Distributor ID:	Raspbian
> Description:	Raspbian GNU/Linux 9.3 (stretch)
> Release:	9.3
> Codename:	stretch

XECDesign, 2019-04-15:
Just ran it on fresh raspbian install and it seems to be building fine so far. It's well past the bootstrap stage.

I haven't seen that problem before, so I'm not sure where to go from here without more information.

AFICT, the path looks fine. /debootstrap should be within the chroot, not the host filesystem.

dernasherbrezon, 2019-04-27:
Looks like an issue with debootstrap. Running on completely new ubuntu xenial (16.04) gives the following:

> root@203267:~/pi-gen#  apt list -a debootstrap
> Listing... Done
> debootstrap/xenial-updates,xenial-updates,now 1.0.78+nmu1ubuntu1.8 all [installed]
> debootstrap/xenial,xenial 1.0.78+nmu1ubuntu1 all

Looks like 1.0.78 is compatible and 1.0.89 is not.

closed 2019-04-27 by dernasherbrezon

dummychunk, 2020-11-22:
The error is caused by the existence of the `/debootstrap` directory on the host system.
Excerpt from `/usr/sbin/debootstrap`:
```
if [ -z "$DEBOOTSTRAP_DIR" ]; then
	if [ -x /debootstrap/debootstrap ]; then
		DEBOOTSTRAP_DIR=/debootstrap
	else
		DEBOOTSTRAP_DIR=/usr/share/debootstrap
	fi
fi```