RPi-Distro/pi-gen #282: ssh

mikysal78, 2019-05-01:
howto for create file authorized_keys for root?
howto change default port ssh?

XECDesign, 2019-05-01:
You can install files like I do here - https://github.com/RPi-Distro/pi-gen/blob/master/stage2/01-sys-tweaks/01-run.sh

And you can drop in patches, like you see here - https://github.com/RPi-Distro/pi-gen/tree/master/stage2/01-sys-tweaks/00-patches

mikysal78, 2019-05-01:
> You can install files like I do here - https://github.com/RPi-Distro/pi-gen/blob/master/stage2/01-sys-tweaks/01-run.sh
> 
ok, but create in the home FIRST_USER dir .ssh with root don'user.
Because?
install -d  "${ROOTFS_DIR}/home/ninux/.ssh"
install -d  "${ROOTFS_DIR}/root/.ssh"
install -m 600 files/authorized_keys2   "${ROOTFS_DIR}/home/ninux/.ssh/"
install -m 600 files/authorized_keys2   "${ROOTFS_DIR}/root/.ssh/"

> And you can drop in patches, like you see here - https://github.com/RPi-Distro/pi-gen/tree/master/stage2/01-sys-tweaks/00-patches
>
this is ok for chage default port on ssh

Thanks.

XECDesign, 2019-05-01:
> ok, but create in the home FIRST_USER dir .ssh with root don'user.
> Because?

Pardon?

mikysal78, 2019-05-01:
ninux@ninuxberry:~ $ ls -la
totale 24
drwxr-xr-x 3 ninux ninux 4096 mag  1 16:39 .
drwxr-xr-x 3 root  root  4096 mag  1 16:23 ..
-rw-r--r-- 1 ninux ninux  220 mag  1 16:23 .bash_logout
-rw-r--r-- 1 ninux ninux 3523 mag  1 16:23 .bashrc
-rw-r--r-- 1 ninux ninux  675 mag  1 16:23 .profile
drwxr-xr-x 2 **root  root**  4096 mag  1 16:39 .ssh
ninux@ninuxberry:~ $

XECDesign, 2019-05-01:
You can specify the permissions/ownership you want - https://github.com/RPi-Distro/pi-gen/blob/master/stage4/03-magpi/00-run.sh#L14

`man install`

closed 2019-05-01 by mikysal78