RPi-Distro/pi-gen #286: Why default to GB keyboard layout?

matthijskooijman, 2019-05-15:
This has been bugging me since I started using Raspbian, but now I think I've found the place where this is defined.

Why is the default keyboard layout the GB layout? It seems better to default to a US layout. I can think of a few reasons:
 - Most operating systems default to US
 - I suspect it might be the most used layout (not just in the US, but also in other countries, such as the Netherlands where I'm based)
 - The GB layout does not seem to have the pipe `|`, which makes operating the shell a challenge.

Would it make sense to change this default? Seems it's a matter of updating `stage2/01-sys-tweaks/00-debconf` (though I haven't tried yet).

drothmaler, 2019-05-15:
Would be even better, if it could be configured, for the build :-)

matthijskooijman, 2019-05-15:
Good plan! One challenge is that it might not be entirely trivial to capture the keyboard configuration in a single value (there's model, layout, variant and settings for alt, compose, ctrl-alt-backspace IIRC), but perhaps it would be good enough to at least be able to select layout and variant (and perhaps also model), maybe colon-separated (I think that's what the keyboard-configuration script uses internally as well).

XECDesign, 2019-05-15:
GB layout because it's a UK product.

`|` is right_alt+`

raspi-config and the first run wizard make changing the layout simple.

There's currently no config option within pi-gen for this, but you can change it here - https://github.com/RPi-Distro/pi-gen/blob/master/stage2/01-sys-tweaks/00-debconf . An easy way to get those options for your keymap is to run `debconf-get-selections | grep keyboard-configuration`.

The debconf-based keymap selection stuff seems to be broken in Buster, so maybe the better approach now is to modify `/etc/default/keyboard` directly rather than via `debconf`. That would also make it simpler to add a pi-gen config option.

matthijskooijman, 2019-05-15:
> GB layout because it's a UK product.
Ah, right. It's gone a bit wider since then, though :-p

> | is right_alt+`
Good to know :-)

> The debconf-based keymap selection stuff seems to be broken in Buster, so maybe the better approach now is to modify /etc/default/keyboard directly rather than via debconf. That would also make it simpler to add a pi-gen config option.

I added a custom stage (on a stretch image) that modifies the keyboard layout after stage2 and also found that it wouldn't process modifications in debconf. I ended up just modifying `/etc/default/keyboard` instead, which I think is where the `keyboard-configuration.config` script actually reads the default values from (so I guess debconf might only be used for defaults at installation time and to communicatie from the config to the postinst script, I think).

In any case, I'll see if I can submit a PR that makes it configurable from pi-gen's `config` file, if you're interested? I'll make sure to also test with buster, thanks for that headsup.

XECDesign, 2019-05-15:
debconf selections added before a package is installed are treated differently than those afterwards.

The idea is that you can 'preseed' values and then when the package is installed it can use those instead of asking the user or choosing its own defaults.

A PR would be good, thank you.

closed 2019-07-12 by XECDesign

matthijskooijman, 2019-10-13:
I was finally getting around to preparing a PR for this, only to find out I was beaten to it by @jbussdieker in https://github.com/RPi-Distro/pi-gen/pull/323 which was already merged as well. Thanks @jbussdieker :-)