RPi-Distro/pi-gen #290: Extra packages not available

wtgee, 2019-06-03:
Hi, I'm trying to add some custom packages and other things to an install and they are simply not appearing once I boot the resulting image.  I can see the items being built during in the log output but there is no trace of these at the end.  I have tried various different things in terms of moving programs to various stages but am confused as to what might be causing it.

For example, I'm trying to auto-install Docker via the get.docker.com url by adding the appropriate scripts to (currently although tried other stages) the `stage3/01-tweaks/00-run.sh` script.  As mentioned, both this and another custom install script (`gphoto2-updater`) build correctly according to the logs. I've tried adding an `EXPORT_IMAGE` file to `stage3` to no avail and originally had this all in `stage5` (then `stage4` and at one point `stage2`).

Anything obvious I am missing?  Thanks!

XECDesign, 2019-06-03:
Could you post a snippet of the run.sh file that you've added?

wtgee, 2019-06-03:
`stage3/01-tweaks/00-run.sh` is as follows:

```bash
!/bin/bash -e

rm -f "${ROOTFS_DIR}/etc/systemd/system/dhcpcd.service.d/wait.conf"

# Install Docker
curl -fsSL https://get.docker.com -o get-docker.sh
sh get-docker.sh

# gphoto2 updater
wget https://raw.githubusercontent.com/gonzalo/gphoto2-updater/master/gphoto2-updater.sh && chmod +x gphoto2-updater.sh && ./gphoto2-updater.sh
```

I watched this build and everything looked ok as it was running. I'm not sure how to find the details on the build log after the fact (I'm using `build-docker.sh`). The most detail I can find is from `deploy/build.log`:

```
[00:00:22] Begin /pi-gen/stage3
[00:00:22] Begin /pi-gen/stage3/prerun.sh
[00:00:27] End /pi-gen/stage3/prerun.sh
[00:00:27] Begin /pi-gen/stage3/00-install-packages
[00:00:27] Begin /pi-gen/stage3/00-install-packages/00-debconf
[00:00:29] End /pi-gen/stage3/00-install-packages/00-debconf
[00:00:29] Begin /pi-gen/stage3/00-install-packages/00-packages-nr
[00:09:46] End /pi-gen/stage3/00-install-packages/00-packages-nr
[00:09:46] Begin /pi-gen/stage3/00-install-packages/00-packages
[00:16:44] End /pi-gen/stage3/00-install-packages/00-packages
[00:16:44] Begin /pi-gen/stage3/00-install-packages/01-run.sh
[00:16:44] End /pi-gen/stage3/00-install-packages/01-run.sh
[00:16:44] End /pi-gen/stage3/00-install-packages
[00:16:44] Begin /pi-gen/stage3/01-tweaks
[00:16:44] Begin /pi-gen/stage3/01-tweaks/00-run.sh
[00:19:29] End /pi-gen/stage3/01-tweaks/00-run.sh
[00:19:29] End /pi-gen/stage3/01-tweaks
[00:19:29] End /pi-gen/stage3
```

I don't really thing `stage3` is the appropriate place but as mentioned I couldn't get it to work anywhere and this is just my latest attempt.

XECDesign, 2019-06-03:
You're running it on the host system.

Look at the use of `on_chroot` in existing run files.

wtgee, 2019-06-03:
Okay, that sort of seems obvious after the fact, but just to make sure I understand, I should be able to just copy those same contents (docker and gphoto2) to `stage3/01-tweaks/00-run-chroot.sh` and have it work?  Trying now, thanks!

XECDesign, 2019-06-03:
I don't think I actually use that particular approach anywhere so I can't vouch for it still working.

What I meant was keep everything as it was, but do it like this:
```
on_chroot << EOF
curl -fsSL https://get.docker.com -o get-docker.sh
sh get-docker.sh

# gphoto2 updater
wget https://raw.githubusercontent.com/gonzalo/gphoto2-updater/master/gphoto2-updater.sh && chmod +x gphoto2-updater.sh && ./gphoto2-updater.sh
EOF
```

wtgee, 2019-06-03:
Ah, got it. Will try and report. Thanks.

kraeml, 2019-06-03:
I run this in stage2:

```
#!/bin/bash -e

on_chroot << EOF
if ! docker -v
then
	echo 'Installing Docker'
	curl -SsL https://get.docker.com | sh
fi
EOF
```

wtgee, 2019-06-03:
Ok, this worked without problems. I'll clean up my end a little and then see if I think a PR for some doc changes is worthwhile. Thanks for the help!

closed 2019-06-03 by wtgee