RPi-Distro/pi-gen #291: Issues when running Chromium

matiasdecarli, 2019-06-12:
I'm trying to create a lightweight image to run Chromium Browser in Kiosk Mode

I added a new stage after `stage2` to install custom tools (the app to display on the kiosk, for example). Also, after this custom stage, I'm ignoring `stage3`, `stage4` and `stage5`

Everything builds OK (I'm using the docker build), but I'm getting an error when trying to start chromium browser. I'm getting the following

```
guest@pi-gen:~$ chromium-browser
 --disable-quic --enable-tcp-fast-open --ppapi-flash-path=/usr/lib/chromium-browser/libpepflashplayer.so --ppapi-flash-args=enable_stagevideo_auto=0 --ppapi-flash-version=
[3409.966902] Unhandled prefetch abort: breakpoint debug exception (0x002) at 0x00abb2d8
[3409.988012] Unhandled prefetch abort: breakpoint debug exception (0x002) at 0x0305cb22
Trace/breakpoint trap
```

Any clues on why this is happening?

XECDesign, 2019-06-13:
Can you reproduce the issue on a standard Raspbian issue? Can you do it on a lite image? You'd need to find the minimum amount of changes required to introduce the problem into an officially released image.

matiasdecarli, 2019-06-13:
I'm downloading Raspbian Sketch Lite to see if it happens on a standard image. I'll let you know!

matiasdecarli, 2019-06-17:
@XECDesign is there any place where I can find what is installed on a base Raspbian Stretch image to compare?

Right now I'm getting a lot of `Hash Sum mismatch` when trying to install Chrome on a fresh new Raspbian Stretch, so I cannot compare

XECDesign, 2019-06-17:
The output of dpkg -l can be found here:
https://downloads.raspberrypi.org/raspbian/images/raspbian-2019-04-09/2019-04-08-raspbian-stretch.info

matiasdecarli, 2019-06-17:
Amazing, will take a look and see if I can fix this. I'll post my findings for anyone running the same issue

matiasdecarli, 2019-06-19:
What I saw is that as soon as I put one more stage after `stage2`... even if the stage just copy files from the host to the new image, this new image gets "broken" (By broken I mean that fails after I install `chromium-browser`)

I came to the point where 2 `dpkg -l` (raspbian lite, and my custom image lite) with same packages, one fails and the other runs OK

Do u think that something other than a package could be breaking my custom stage?

Thanks in advance, I'ts been really helpfull

XECDesign, 2019-06-20:
Just `EXPORT_IMAGE` is enough. Did you change the content of `EXPORT_IMAGE`? Do you have a branch somewhere with your changes I can take a look at?

matiasdecarli, 2019-06-20:
The `EXPORT_IMAGE` it's the same file as the one on `stage2`.

 Also, I moved every folder on that custom stage that I have, to be on the original `stage2`... sill failing. So, I believe is not something related with the custom stage that i'm creating. 

I'm inviting you to a private repository. You can check the branch #development. That's the one that's failing. That branch finishes succesfully but when I run `chromium-browser` I get the `Trace/breakpoint trap`

Thanks again for your help

matiasdecarli, 2019-06-26:
I think it could have been related with some specific version of Chromium, because now is working.

Thanks a lot @XECDesign , I've learned a lot in the process of debugging

closed 2019-06-26 by matiasdecarli

XECDesign, 2019-06-27:
Sorry about disappearing. It has been a busy week. Glad to hear it's resolved now.

matiasdecarli, 2019-07-09:
> Sorry about disappearing. It has been a busy week. Glad to hear it's resolved now.

No problem, you helped me a lot!