RPi-Distro/pi-gen #293: Pull docker image while building image

p3root, 2019-06-18:
Hey!

I would like to now if I could pull a docker image and cache that image while building the rasbpian? I have tried several things without success. Pulling the image on first startup is working, but not what I want to achive.

I have found 2 docker forum articles but those are unanswered. 

* https://forums.docker.com/t/possible-to-install-docker-images-with-debootstrap/30576
* https://forums.docker.com/t/is-it-possible-to-pull-an-docker-image-without-the-engine/29961

Thanks in advance

XECDesign, 2019-06-18:
Not sure what you mean. Instead of debootstrapping? What would be the advantage that you couldn't get without docker?

matiasdecarli, 2019-06-19:
Hey @p3root I think that I know what you're trying to do (because I'm doing the same thing)

If you're trying to download some images and have them ready when you first boot, you can use this docker functions:

https://docs.docker.com/engine/reference/commandline/save/
https://docs.docker.com/engine/reference/commandline/load/

This way you can save the images as a `.tar` and pass them to the OS on any stage that you want. Then, on your first run, you can do a `docker run` with the downloaded images

p3root, 2019-06-19:
Hey @matiasdecarli 

I already found that solution. Also I found this one:

https://raw.githubusercontent.com/moby/moby/master/contrib/download-frozen-image-v2.sh
from here
https://devops.stackexchange.com/questions/2731/downloading-docker-images-from-docker-hub-without-using-docker

But you also need some time to bootstrap that image on first startup. I want to be ready-to-go on my first boot. 

Any other solutions out there?

cheers,
Patrik

XECDesign, 2019-07-23:
Okay to close?

p3root, 2019-07-23:
Thanks anyway :)

closed 2019-07-23 by p3root

Servuc, 2019-11-20:
This issue is closed but I'm on the same way.

We have a startup script that unpack with `docker load` but it's ugly and slow.
If someone has an idea ? 

Have a good day :)

mayankchhabra, 2020-07-04:
I know this is an old issue but curious to know if anyone could get it working? `docker save` and `docker load` works, but like @Servuc mentioned, it is ugly and slow.

Servuc, 2020-07-06:
Hey,

It's old, but I thougth that I have give more information. As @mayankchhabra said, `docker save/load`, it's very ugly ! 

We have found another way, ugly, but without an eternal loading at first startup,

We make a copy from some dirs on another RPi, that just contains file for containers, put all in tar. And extract that tar in OS building in CI. It's working all the time :) 

I'm looking for that work to give you more details ;)

++

mayankchhabra, 2020-07-06:
@Servuc that's similar to we just found out too :) 

We're copying the `/var/lib/docker/` directory from the host to the filesystem of image and it seems to be working great! 

Images on the host are fetched with `docker pull --platform=linux/arm/v7`

Servuc, 2020-07-07:
Yes we do that ^^'
We don't found a better way

I don't think there is a normal way. Theorically, you couldn't start Docker in Docker. You can access Docker host from container (DinD) but not a Docker instance into another Docker instance.

cdupont, 2021-09-13:
Copying the `/var/lib/docker/` directory from the host doesn't seem so clean, that folder might contain other images...

Servuc, 2021-09-13:
Yes, i agree. But you can make that folder in a CI context same as first use :)

ivelin, 2021-09-13:
Just noting that #470 is a duplicate of this issue with additional context.
Thanks @cdupont for connecting the two issues.

cdupont, 2021-09-14:
See also this question: https://superuser.com/questions/1547606/pulling-an-docker-image-within-a-chroot-environment