RPi-Distro/pi-gen #294: Builds fail on RPi 3B+ with Buster

dlabun, 2019-06-26:
I pulled the latest copy of the pi-gen project and builds are not completing due to missing file / folder issues. I assume this must be related to the move to Buster. This was run on a RPi 3B+ with a clean copy of Raspbian I just downloaded from the RPi website.

`pi@raspberrypi:~/pi-gen-master $ sudo bash build.sh
[11:43:34] Begin /home/pi/pi-gen-master
[11:43:34] Begin /home/pi/pi-gen-master/stage0
[11:43:34] Begin /home/pi/pi-gen-master/stage0/prerun.sh
I: Target architecture can be executed
I: Retrieving InRelease 
I: Checking Release signature
I: Valid Release signature (key id A0DA38D0D76E8B5D638872819165938D90FDDD2E)
I: Retrieving Packages 
I: Validating Packages 
I: Retrieving Packages 
I: Validating Packages 
I: Retrieving Packages 
I: Validating Packages 
I: Resolving dependencies of required packages...
I: Resolving dependencies of base packages...
I: Checking component main on http://raspbian.raspberrypi.org/raspbian...
I: Retrieving adduser 3.118
I: Validating adduser 3.118
I: Retrieving apt 1.8.2
I: Validating apt 1.8.2
I: Retrieving apt-utils 1.8.2
I: Validating apt-utils 1.8.2
I: Retrieving base-files 10.3+rpi1
I: Validating base-files 10.3+rpi1
I: Retrieving base-passwd 3.5.46
I: Validating base-passwd 3.5.46
I: Retrieving bash 5.0-4
I: Validating bash 5.0-4
I: Retrieving bsdmainutils 11.1.2
I: Validating bsdmainutils 11.1.2
I: Retrieving bsdutils 1:2.33.1-0.1
I: Validating bsdutils 1:2.33.1-0.1
I: Retrieving coreutils 8.30-3
I: Validating coreutils 8.30-3
I: Retrieving cpio 2.12+dfsg-9
I: Validating cpio 2.12+dfsg-9
I: Retrieving cron 3.0pl1-134
I: Validating cron 3.0pl1-134
I: Retrieving dash 0.5.10.2-5
I: Validating dash 0.5.10.2-5
I: Retrieving debconf 1.5.71
I: Validating debconf 1.5.71
I: Retrieving debconf-i18n 1.5.71
I: Validating debconf-i18n 1.5.71
I: Retrieving debianutils 4.8.6.1
I: Validating debianutils 4.8.6.1
I: Retrieving diffutils 1:3.7-3
I: Validating diffutils 1:3.7-3
I: Retrieving dirmngr 2.2.12-1+rpi1
I: Validating dirmngr 2.2.12-1+rpi1
I: Retrieving dmidecode 3.2-1
I: Validating dmidecode 3.2-1
I: Retrieving dmsetup 2:1.02.155-3
I: Validating dmsetup 2:1.02.155-3
I: Retrieving dpkg 1.19.7
I: Validating dpkg 1.19.7
I: Retrieving e2fsprogs 1.44.5-1
I: Validating e2fsprogs 1.44.5-1
I: Retrieving fdisk 2.33.1-0.1
I: Validating fdisk 2.33.1-0.1
I: Retrieving findutils 4.6.0+git+20190209-2
I: Validating findutils 4.6.0+git+20190209-2
I: Retrieving gcc-4.9-base 4.9.4-2+rpi1+b19
I: Validating gcc-4.9-base 4.9.4-2+rpi1+b19
I: Retrieving gcc-5-base 5.5.0-8
I: Validating gcc-5-base 5.5.0-8
I: Retrieving gcc-6-base 6.5.0-1+rpi1+b1
[... 554 lines cut ...]
[11:45:42] End /home/pi/pi-gen-master/stage3/00-install-packages/01-run.sh
[11:45:42] End /home/pi/pi-gen-master/stage3/00-install-packages
[11:45:42] Begin /home/pi/pi-gen-master/stage3/01-tweaks
[11:45:42] Begin /home/pi/pi-gen-master/stage3/01-tweaks/00-run.sh
[11:45:42] End /home/pi/pi-gen-master/stage3/01-tweaks/00-run.sh
[11:45:42] End /home/pi/pi-gen-master/stage3/01-tweaks
[11:45:42] End /home/pi/pi-gen-master/stage3
[11:45:42] Begin /home/pi/pi-gen-master/stage4
[11:45:42] Begin /home/pi/pi-gen-master/stage4/prerun.sh
[11:45:43] End /home/pi/pi-gen-master/stage4/prerun.sh
[11:45:45] Begin /home/pi/pi-gen-master/stage4/00-install-packages
[11:45:45] Begin /home/pi/pi-gen-master/stage4/00-install-packages/00-debconf
mount: /home/pi/pi-gen-master/work/2019-06-26-Raspbian/stage4/rootfs/proc: mount point does not exist.
mount: /home/pi/pi-gen-master/work/2019-06-26-Raspbian/stage4/rootfs/dev: mount point does not exist.
realpath: /home/pi/pi-gen-master/work/2019-06-26-Raspbian/stage4/rootfs/dev/pts: No such file or directory
mount: /home/pi/pi-gen-master/work/2019-06-26-Raspbian/stage4/rootfs/sys: mount point does not exist.
execve /bin/bash failed!
[11:45:46] End /home/pi/pi-gen-master/stage4/00-install-packages/00-debconf
[11:45:46] Begin /home/pi/pi-gen-master/stage4/00-install-packages/00-packages-nr
mount: /home/pi/pi-gen-master/work/2019-06-26-Raspbian/stage4/rootfs/proc: mount point does not exist.
mount: /home/pi/pi-gen-master/work/2019-06-26-Raspbian/stage4/rootfs/dev: mount point does not exist.
realpath: /home/pi/pi-gen-master/work/2019-06-26-Raspbian/stage4/rootfs/dev/pts: No such file or directory
^C
pi@raspberrypi:~/pi-gen-master $ 
`

XECDesign, 2019-06-27:
Not sure why your build continues. Mine fails with bootstrap.

If I recall correctly, debootstrap doesn't like running through capsh. Since it's not important during the bootstrap stage, we can just remove it, but it would be nice to know what the actual issue is.

artynet, 2019-06-27:
I also got this error when using the docker build....

```
[22:19:29] Begin /pi-gen/stage1/02-net-tweaks/00-packages
Reading package lists...
Building dependency tree...
Reading state information...
netbase is already the newest version (5.6).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
[22:19:30] End /pi-gen/stage1/02-net-tweaks/00-packages
[22:19:30] Begin /pi-gen/stage1/02-net-tweaks/00-patches
The quilt meta-data in .pc/ are already in the version 2 format; nothing to do
File series fully applied, ends at patch /pi-gen/stage1/02-net-tweaks/00-patches/01-hosts.diff
[22:19:30] End /pi-gen/stage1/02-net-tweaks/00-patches
[22:19:30] Begin /pi-gen/stage1/02-net-tweaks/00-run.sh
[22:19:30] End /pi-gen/stage1/02-net-tweaks/00-run.sh
[22:19:30] End /pi-gen/stage1/02-net-tweaks
[22:19:30] Begin /pi-gen/stage1/03-install-packages
[22:19:30] Begin /pi-gen/stage1/03-install-packages/00-packages
Reading package lists...
Building dependency tree...
Reading state information...
libraspberrypi-bin is already the newest version (1.20190620+1-1).
libraspberrypi0 is already the newest version (1.20190620+1-1).
raspi-config is already the newest version (20190620).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
[22:19:31] End /pi-gen/stage1/03-install-packages/00-packages
[22:19:31] End /pi-gen/stage1/03-install-packages
[22:19:31] End /pi-gen/stage1
[22:19:31] Begin /pi-gen/stage2
[22:19:31] Begin /pi-gen/stage2/prerun.sh
[22:19:31] End /pi-gen/stage2/prerun.sh
[22:19:31] Begin /pi-gen/stage2/00-copies-and-fills
[22:19:31] Begin /pi-gen/stage2/00-copies-and-fills/01-packages
Reading package lists...
Building dependency tree...
Reading state information...
raspi-copies-and-fills is already the newest version (0.13).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
[22:19:33] End /pi-gen/stage2/00-copies-and-fills/01-packages
[22:19:33] Begin /pi-gen/stage2/00-copies-and-fills/02-run.sh
mv: cannot stat '/pi-gen/work/2019-06-26-raspbian-nj/stage2/rootfs/etc/ld.so.preload': No such file or directory

real    0m18,773s
user    0m0,065s
sys     0m0,018s
```

dlabun, 2019-06-29:
For the sake of troubleshooting I pulled a clean copy of Pi-Gen to a 3B+ running Stretch and building Buster worked fine. The problem must reside within a change made in Debian between Stretch and Buster and not Pi-Gen itself.

RussellJoyce, 2019-07-03:
> If I recall correctly, debootstrap doesn't like running through capsh. Since it's not important during the bootstrap stage, we can just remove it, but it would be nice to know what the actual issue is.

`capsh` doesn't appear to be the issue, or at least replacing `capsh --drop=cap_setfcap --` with `/bin/bash` in `scripts/common` still causes the build to fail for me, at the "rmdir: failed to remove ... Directory not empty" error (on latest Raspbian Buster).

benp44, 2019-07-10:
> `capsh` doesn't appear to be the issue, or at least replacing `capsh --drop=cap_setfcap --` with `/bin/bash` in `scripts/common` still causes the build to fail for me, at the "rmdir: failed to [...]

The problem definitely seemed to be capsh for me. The build seems to be working since I replaced 
```
setarch linux32 capsh --drop=cap_setfcap -- "${BOOTSTRAP_CMD}" "${BOOTSTRAP_ARGS[@]}" || true
```
with just
```
debootstrap "${BOOTSTRAP_ARGS[@]}" || true
```

RussellJoyce, 2019-07-10:
Hmm yep, that works for me too, thanks.
Looks like my mistake was thinking that replacing it with `/bin/bash` would be a good idea (based on the "Execute /bin/bash with trailing arguments" line in the capsh manpage).

dlabun, 2019-07-10:
It looks like this workaround breaks the APT proxy option, After making the same change on my end I started getting a lot of messages reading "Unsupported proxy configured: 192.168.2.1717//3142". It's odd because I can see Pi-Gen is pulling some files from the proxy at first but then suddenly it doesn't like the proxy any more.

XECDesign, 2019-07-24:
Not working through bash was a good hint. Looks like it runs debootstrap through bash rather than sh then has compatibility issues?

Not sure how to fix this properly. We need capsh because otherwise you end up with files with capabilities set which will have problems on NFS filesystems or when archived with tar.

Capsh will only use bash, but will parse arguments, so ` -c "debootstrap ..."` works. Then things get hairy when it comes to making sure arguments are passed properly.

smg-bg, 2019-09-01:
@jvonau I've tried your change:

#setarch linux32 capsh --drop=cap_setfcap -- "${BOOTSTRAP_CMD}" "${BOOTSTRAP_ARGS[@]}" || true
  debootstrap "${BOOTSTRAP_ARGS[@]}" || true

...and the build seems to work. However, I'm not sure I understand what would be the impact of this change? Could you please clarify?

jvonau, 2019-09-01:
@smg-bg Not sure of the total impact, haven't tried booting the resulting image yet.  I was looking for quick way to continue working towards a custom image farther up the chain and the above mentioned fix got me past the error I was seeing. Easy to revert when there is a proper fix.

dlabun, 2019-09-01:
All of the images I've built with that workaround seem to work fine.

jvonau, 2019-09-01:
Good to know, thanks

smg-bg, 2019-09-01:
I can also confirm that my custom image work with that fix.

XECDesign, 2019-09-02:
If you don't drop `cap_setfcap` you end up with files with capabilities set, which NFS doesn't support. So, if you're netbooting, you may find ping won't work without `sudo`.

XECDesign, 2019-10-28:
I think I have a fix here https://github.com/RPi-Distro/pi-gen/issues/338#issuecomment-546995455

dlabun, 2019-10-28:
I’ll test it this weekend.

closed 2019-10-30 by XECDesign

mutability, 2019-11-15:
An alternative fix here is to pass the full path to debootstrap. (The underlying issue is that in the case that breaks, `$0` within debootstrap is not a fully qualified path)

jerrychong25, 2022-04-15:
Hi @benp44 / @RussellJoyce / @dlabun / @smg-bg ,

Do u guys still encounter same error **"stage0/rootfs/proc: mount point does not exist."** recently? If yes, can share your solution on this issue?

As I encounter same error today, but latest fix by @XECDesign on 29th October 2019 is already in the **scripts/common** file.

Here is error screenshot:
[image]

Here is scripts/common file code:
[image]

Thanks in advance!

commit 55d41b1c (2019-10-30): Ensure debootstrap runs through 'sh' rathern than 'bash'

This fixes #338 and fixes #294
commit 19c01887 (2019-10-30): Ensure debootstrap runs through 'sh' rather than 'bash'

This fixes #338 and fixes #294