RPi-Distro/pi-gen #297: Issue With SSL Certificate Verification

ryanteck, 2019-06-27:
We use pi-gen to produce images for our products. However after upgrading our fork to buster seem to be experiencing an issue with open SSL  in PHP failing to verify SSL Certificates.

Once the Pi has booted if I run the following the issue will resolve. 

`
sudo apt purge ca-certificates 
sudo apt install ca-certificates
`

However I'm not sure why this is happening in the first place as before the buster upgrade didn't have any issues at all.

hhromic, 2019-06-27:
This is a known problem. See #271 
You will need to build using a 32 bits host for now, for example under a virtual machine.

ryanteck, 2019-06-27:
When will this be fixed as it kind of breaks my entire workflow having to use virtual machines. (I have a build server setup in the garage which automates builds with jenkins).

For now I've been able to get around it however guess there would then be more issues then?

ryanteck, 2019-06-27:
Hmm, we can use a Raspberry Pi to build these images on is what I'm reading in #271? 

Looks like the best route to do, I'll report back.

hhromic, 2019-06-27:
It's kind of a complex bug to solve as it seems that a kernel patch is also necessary.
This is by no means the fault of `pi-gen`, so nothing can be done here.

Yes you can use a Raspberry Pi to build too (as the Foundation is doing) which doesn't require Qemu, therefore avoiding the problem completely. However I'm not sure how much slower is this compared to building in an x86 machine. You don't need to use virtual machines, just a 32 bits host system (Debian Buster recommended).

ryanteck, 2019-06-27:
Unfortunately I want to use more than 4GB of my Ram so changing to a 32 Bit system in 2019 is hardly practical. I'll sort something out.

hhromic, 2019-07-06:
@ryanteck see PR #307 
It was proven possible to build working images with Docker under 64 bits host OS.
You don't need to use 32-bits VMs nor install 32-bits linux. You can give it a try.

ryanteck, 2019-07-06:
Thanks. To be honest I have quite a bit of hassle with docker so its easier for me to VM.  But thanks for letting me know!

closed 2019-07-06 by ryanteck